ome/bioformats #1515: "Cannot construct image with 7 channels" error on trying to convert some TIFFs to JP2

axfelix, 2015-01-07:
Hi folks,

I have some questionable tiffs that I've been getting from old imaging equipment that I've been using Djatoka to convert up to JP2 up to now. I do not have much love for Djatoka, and was very excited when I discovered this project, as it means I can potentially convert a much wide range of file formats to more preservation-friendly standards. However, I'm getting the following Java error when trying to run bfconvert on some of these tiffs:

axfelix@shoebox:/opt/bftools$ bfconvert -merge -expand /opt/image.tif ~/Desktop/test.jp2
/opt/image.tif
TiffDelegateReader initializing /opt/image.tif
Reading IFDs
Populating metadata
Checking comment style
Populating OME metadata
[Tagged Image File Format] -> /home/axfelix/Desktop/test.jp2 [JPEG-2000]
Exception in thread "main" java.lang.IllegalArgumentException: Cannot construct image with 7 channels
	at loci.formats.gui.AWTImageTools.constructImage(AWTImageTools.java:659)
	at loci.formats.codec.JPEG2000Codec.compress(JPEG2000Codec.java:137)
	at loci.formats.out.JPEG2000Writer.compressBuffer(JPEG2000Writer.java:126)
	at loci.formats.out.JPEG2000Writer.saveBytes(JPEG2000Writer.java:86)
	at loci.formats.FormatWriter.saveBytes(FormatWriter.java:126)
	at loci.formats.ImageWriter.saveBytes(ImageWriter.java:201)
	at loci.formats.tools.ImageConverter.convertPlane(ImageConverter.java:547)
	at loci.formats.tools.ImageConverter.testConvert(ImageConverter.java:483)
	at loci.formats.tools.ImageConverter.main(ImageConverter.java:677)

When trying to convert the same image to jpeg rather than jp2, I get: Exception in thread "main" javax.imageio.IIOException: Invalid argument to native writeImage, with the same traceback.

I can get both of these to work by using the -separate flag (thanks for the good documentation that helped me figure that out), but unfortunately, that results in greyscale output.

Any ideas? Thanks!

melissalinkert, 2015-01-08:
If the original TIFF has more than 4 color channels, that error is expected - JPEG and JPEG-2000 files won't be able to store that many channels as a single image.  Converting to OME-TIFF with JPEG-2000 compression would be one option, as that would compress the pixel data in a single file while preserving the metadata.  ```bfconvert -separate /opt/image.tif -compression JPEG-2000 test.ome.tiff``` should work.

If the original TIFF is not expected to have 7 channels, then that's something we'll need to fix in the reader.  If you are willing to upload a file to https://www.openmicroscopy.org/qa2/qa/upload/, then we can take a look at what's going wrong.

axfelix, 2015-01-09:
It's very plausible that the original TIFF has 7 channels. Would it make sense for me to try to run it through Imagemagick beforehand? I'm used to munging images that way, and I'm assuming that Imagemagick has some way of silently consolidating images into 3/4 color channels, since this hasn't come up before. Failing that, would it be plausible for OME-TIFF to add a flag that re-rasterizes an image down to a reasonable RGB equivalent, so that the only option isn't greyscale output? Sorry if I'm making too many assumptions -- just curious.

melissalinkert, 2015-01-12:
The ```-composite``` option to Imagemagick's ```convert``` command may be useful, but I don't know if that will work for this type of data.  If that doesn't work, ImageJ should be able to do what you want.

If you install ImageJ and the Bio-Formats plugin per https://www.openmicroscopy.org/site/support/bio-formats5/users/imagej/index.html, then the following steps should result in a single RGB JPEG-2000 file:

- start ImageJ
- select "Plugins > Bio-Formats > Bio-Formats Importer"
- choose the input file
- change the "Color mode" option to "Composite" and select the "Autoscale" option in the "Bio-Formats Import Options" window (all other options can be left default)
- select "Image > Color > Stack to RGB"
- select "Plugins > Bio-Formats > Bio-Formats Exporter" to save the image

If you're processing a lot of files, this can be sped up using a macro:

```
run("Bio-Formats Importer", "autoscale color_mode=Composite view=Hyperstack stack_order=Default");
run("Stack to RGB");
run("Bio-Formats Exporter", "compression=[JPEG-2000 Lossy]");
```

Pasting the macro into a file and selecting "Plugins > Macros > Run..." does the same as the above, but only prompts for input and output files.

axfelix, 2015-01-12:
Thanks, Melissa! My interest is primarily in being able to do this via cli jobs, so ImageJ isn't too useful for me, but this has all been really helpful.

axfelix, 2015-01-12:
Oh, and just to confirm -- batch importing and exporting images via OMERO utilizes the same libraries as bfconvert (so there's no need to run an entire OMERO instance if this is my primary interest), right?

closed 2015-01-12 by axfelix

melissalinkert, 2015-01-12:
Yes, OMERO uses Bio-Formats for import/export, so there isn't much value in using it to replace bfconvert.