ome/bioformats #1593: JpegXR support, OME ticket #8493

Jondeen, 2015-02-12:
Hi!

http://trac.openmicroscopy.org.uk/ome/ticket/8493

I see the ticket mentions implementing this functionality in beta releases -- are these available as of yet? I cannot find any code updates relevant to this functionality in any of the OME GIT repositories.

Furthermore - is there anything I can provide of code or examples that could help you get this along in any way?

Sincerely,
Jonas Øgaard

melissalinkert, 2015-02-14:
Some preliminary investigation was performed here: https://github.com/openmicroscopy/bioformats/tree/develop/components/ome-jxr

but that code is not currently functional, and is not being actively developed at the moment.  We have plenty of examples of JPEG-XR data, but do not currently have sufficient time to implement complete support for it.  We will update the ticket when a timeline is available and as progress is made; if you would like to be CC'd on the ticket, just let us know.

melissalinkert, 2015-02-23:
Closing, since the JPEG-XR status is noted on Trac.  @Jondeen, please do let us know if you would like to be notified of changes to the above-mentioned ticket.

closed 2015-02-23 by melissalinkert