ome/bioformats #1719: Error on opening CZI file -- bfGetPlane

gvoysey, 2015-04-08:
I'm attempting to open a 760 MB CZI file acquired by another research group with, i believe, Zen Lite.   I have installed the matlab toolbox, added it to my path, and set my java maximum heap memory to 1024 MB.  I'm running r2012a (64 bit) on Windows 7. 

Executing 
``` matlab
 data = bfopen('IP-Stitching-01-Change Scaling-01.czi');
```

results in 
``` java
ZeissCZIReader initializing C:\Projects\LMR-Acquisition\IP-Stitching-01-Change Scaling-01.czi

Unknown IlluminationType value 'Fluorescence' will be stored as "Other"

ome.xml.model.Channel@66156d8 reference to null missing from object hierarchy.

ome.xml.model.Channel@28b689e0 reference to null missing from object hierarchy.

ome.xml.model.Channel@7b0064e3 reference to null missing from object hierarchy.

ome.xml.model.Channel@26b72884 reference to null missing from object hierarchy.

ome.xml.model.Channel@232a32bf reference to null missing from object hierarchy.

ome.xml.model.Channel@150abd60 reference to null missing from object hierarchy.

ome.xml.model.Channel@1a15deb6 reference to null missing from object hierarchy.

ome.xml.model.Channel@a756b37 reference to null missing from object hierarchy.

ome.xml.model.Channel@45b6867 reference to null missing from object hierarchy.

ome.xml.model.Channel@75af8109 reference to null missing from object hierarchy.

ome.xml.model.Channel@782a519b reference to null missing from object hierarchy.

ome.xml.model.Channel@76792357 reference to null missing from object hierarchy.

ome.xml.model.Channel@129e49c0 reference to null missing from object hierarchy.

ome.xml.model.Channel@2db45934 reference to null missing from object hierarchy.

ome.xml.model.Channel@3a78cbab reference to null missing from object hierarchy.

ome.xml.model.Channel@f0896b1 reference to null missing from object hierarchy.

ome.xml.model.Channel@1a0d8377 reference to null missing from object hierarchy.

ome.xml.model.Channel@cabe02e reference to null missing from object hierarchy.

ome.xml.model.Channel@a832ce5 reference to null missing from object hierarchy.

ome.xml.model.Channel@627f7051 reference to null missing from object hierarchy.

Reading series #1
    .Error using loci.formats.ChannelSeparator/openBytes
Java exception occurred:
loci.formats.FormatException: Buffer too small (got 18172584, expected 54517752).

	at loci.formats.FormatTools.checkBufferSize(FormatTools.java:976)

	at loci.formats.FormatTools.checkPlaneParameters(FormatTools.java:932)

	at loci.formats.in.ZeissCZIReader.openBytes(ZeissCZIReader.java:292)

	at loci.formats.ImageReader.openBytes(ImageReader.java:453)

	at loci.formats.ChannelFiller.openBytes(ChannelFiller.java:156)

	at loci.formats.ChannelSeparator.openBytes(ChannelSeparator.java:225)

	at loci.formats.ChannelSeparator.openBytes(ChannelSeparator.java:157)

Error in bfGetPlane (line 78)
plane = r.openBytes(...

Error in bfopen (line 148)
        arr = bfGetPlane(r, i, varargin{:});
```

melissalinkert, 2015-04-08:
Thanks for reporting this, @gvoysey.  Are you using Bio-Formats 5.1.0?, or an older version?

gvoysey, 2015-04-09:
I downloaded `http://downloads.openmicroscopy.org/bio-formats/5.1.0/artifacts/bfmatlab.zip` , so I'm assuming that it's 5.1.0.  

Incidentally, running `bfUpgradeCheck` returns
```java
Undefined variable "loci" or class "loci.formats.UpgradeChecker".

Error in bfUpgradeCheck (line 44)
upgrader = loci.formats.UpgradeChecker();
```

sbesson, 2015-04-09:
@gvoysey: the error above is due to the absence of the JAR in the Java class path. Can you try running `bfCheckJavaPath` before `bfUpgradeCheck`?

gvoysey, 2015-04-09:
That resolved the `bfUpgradeCheck` error, it now returns `*** bioformats_package.jar is up-to-date ***`. 

The original error in `bfopen` is unchanged.

melissalinkert, 2015-04-12:
OK, thanks for verifying that 5.1.0 is installed.  I haven't been able to reproduce the error message though, unfortunately - would you be willing to upload the problematic file to https://www.openmicroscopy.org/qa2/qa/upload/ (you may need to zip the file first)?

gvoysey, 2015-04-12:
that's fine by me; it's ~800 MB.

gvoysey, 2015-04-13:
@melissalinkert  -- uploaded last night. Have you gotten the file?

melissalinkert, 2015-04-13:
@gvoysey: yes, we received the file - thank you.  https://github.com/openmicroscopy/bioformats/pull/1733 is the proposed fix for this issue; I expect this to be included in the 5.1.1 version of Bio-Formats, which should be available within a couple of weeks.

closed 2015-04-23 by melissalinkert