ome/bioformats #1782: Metadata skew in Bioformats for LightsheetZ1 CZI data

StephanPreibisch, 2015-05-06:
Hi,

I realized that different versions of Bioformats store the individual stack size for the views under different tags. I implemented a workaround here https://github.com/bigdataviewer/SPIM_Registration/commit/d10db40f989c20ec07b3d615f4a2965e20c573d2, but it seems that this is not right.

In the current Fiji there is Bioformats-5.1.1. and it stores the z-stack size metadata under the key:
SizeZ|View|V|Image|Information #1
SizeZ|View|V|Image|Information #2
SizeZ|View|V|Image|Information #3
...

In Eclipse I still use Bioformats-5.1.0 where the key is
Information|Image|V|View|SizeZ #1
Information|Image|V|View|SizeZ #2
Information|Image|V|View|SizeZ #3
...

Or is this change on purpose?

Thanks a lot, Stephan

melissalinkert, 2015-05-06:
Thanks for point this out, @StephanPreibisch.  This was not intentional; the proposed fix is in gh-1784.

melissalinkert, 2015-05-20:
Closing, as the corresponding fix has been tested and merged for 5.1.2.

closed 2015-05-20 by melissalinkert