ome/bioformats #1785: DeltavisionReader trying to open a FEI Tiff file.

avandecreme, 2015-05-06:
Hello,

When running ImageInfo on a particular FEI Tiff file, it crashes in the DeltavisionReader class:
```
Exception in thread "main" java.lang.NegativeArraySizeException
	at loci.formats.in.DeltavisionReader.initPixels(DeltavisionReader.java:387)
	at loci.formats.in.DeltavisionReader.initFile(DeltavisionReader.java:267)
	at loci.formats.FormatReader.setId(FormatReader.java:1395)
	at loci.formats.ImageReader.setId(ImageReader.java:835)
	at loci.formats.ReaderWrapper.setId(ReaderWrapper.java:651)
	at ImageInfo.testRead(ImageInfo.java:989)
	at ImageInfo.main(ImageInfo.java:1071)
```

It appears that the `isThisType(RandomAccessInputStream stream)` method returns true because DV_MAGIC_BYTES_1 is stored at offset 96.
Could this method be made more restrictive by checking other bytes as well?
For example in my case, checking that sizeX, sizeY, imageCount > 0 or that pixelType is between 0 and 6 would have detected that my file is not a Deltavision one.

melissalinkert, 2015-05-07:
Thank you for reporting this, @avandecreme.

Would you be willing to upload the FEI TIFF to https://www.openmicroscopy.org/qa2/qa/upload/?  That will help us to test the fix and ensure that future versions of Bio-Formats do not break this file again.

avandecreme, 2015-05-07:
I tried to upload a sample file. I am not sure whether it worked or not.
Here is the feedback link:  https://www.openmicroscopy.org/qa2/qa2/qa/feedback/11061/?token=ae2a7c48b3d4e2038f1f00541f929411

Note that I had to randomize most of the pixels because I don't think I am allowed to share the original image.

melissalinkert, 2015-05-08:
Thank you for uploading a sample file.  gh-1788 should resolve the problem; I would expect this to be included in the upcoming 5.1.2 release.

melissalinkert, 2015-05-20:
Closing, as the corresponding fix has been tested and merged for 5.1.2.

closed 2015-05-20 by melissalinkert

avandecreme, 2015-05-20:
Yes it works great. Thanks.