ome/bioformats #1824: TJDecompressor.java:305 can cause error in java runtime environment

kkjaergaard, 2015-06-01:
I'm using Bio-Formats 5.1.2, and on converting the sample file below, I get an error message like this:

```
#
# A fatal error has been detected by the Java Runtime Environment:
#
#  SIGSEGV (0xb) at pc=0x00007f0292a96fc1, pid=8072, tid=139649635669760
#
# JRE version: Java(TM) SE Runtime Environment (8.0_45-b14) (build 1.8.0_45-b14)
# Java VM: Java HotSpot(TM) 64-Bit Server VM (25.45-b02 mixed mode linux-amd64 compressed oops)
# Problematic frame:
# C  [libturbojpeg6772939230623658346.so+0x3cfc1]
#
# Failed to write core dump. Core dumps have been disabled. To enable core dumping, try "ulimit -c unlimited" before starting Java again
#
# An error report file with more information is saved as:
# /home/kristian/Playground/omezoomify/sandbox/hs_err_pid8072.log
#
```

**Command executed:** $ bfconvert -compression JPEG -series 0 6_1_Pancreas_HE.ndpi out.tiff

The error occurs in [TJDecompressor.java at line 305](https://github.com/openmicroscopy/bioformats/blob/v5.1.2/components/forks/turbojpeg/src/org/libjpegturbo/turbojpeg/TJDecompressor.java#L305), and I'm not able to locate the error any further than this.

**Sample files:** [6_1_Pancreas_HE.ndpi that causes the error](http://micro.sdu.dk/bio-formats/6_1_Pancreas_HE.ndpi), [AH09.ndpi](http://micro.sdu.dk/bio-formats/AH09.ndpi) and [AH74.ndpi](http://micro.sdu.dk/bio-formats/AH74.ndpi) that works just fine. [Full error log in hs_err_pid8072](http://micro.sdu.dk/bio-formats/hs_err_pid8072.log) is also available.

Please let me know if there's a more appropriate way to report errors - or if this is the wrong audience :)

hflynn, 2015-06-01:
Hi! We recommend people report issues via our mailing list -http://lists.openmicroscopy.org.uk/mailman/listinfo/ome-users - so that the wider community can benefit. Thanks for linking us to the files and log though.

Note that there is likely to be a delay in investigating this for you at the moment because our tech team are en route to Paris for our Annual User Meeting. Someone will get back to you when they can though.

melissalinkert, 2015-06-16:
Thanks, @kkjaergaard, and our apologies for the delay.

I can definitely reproduce the problem; we're looking into a fix now and will update here once we know more.