ome/bioformats #1840: NumberFormatException on CZI Read

derek-r, 2015-06-17:
Hi all,

I'm working with a team who are encountering problems loading Zeiss CZI files with the Bio-Formats toolbox.  These import errors occur both in MATLAB and ImageJ.  The errors occur at least on v5.0.8 and v5.1.2.  The following MATLAB exception is encountered on load:

    Error using bfGetReader (line 89)
    Java exception occurred:
    java.lang.NumberFormatException: For input string: "with length indication"
     
                    at java.lang.NumberFormatException.forInputString(Unknown Source)
     
                    at java.lang.Integer.parseInt(Unknown Source)
     
                    at java.lang.Integer.parseInt(Unknown Source)
     
                    at loci.formats.in.ZeissCZIReader.initFile(ZeissCZIReader.java:515)
     
                    at loci.formats.in.ZeissCZIReader.initFile(ZeissCZIReader.java:478)
     
                    at loci.formats.FormatReader.setId(FormatReader.java:1317)
     
                    at loci.formats.ImageReader.setId(ImageReader.java:753)
     
                    at loci.formats.ReaderWrapper.setId(ReaderWrapper.java:569)
     
                    at loci.formats.ChannelFiller.setId(ChannelFiller.java:259)
     
                    at loci.formats.ReaderWrapper.setId(ReaderWrapper.java:569)
     
                    at loci.formats.ChannelSeparator.setId(ChannelSeparator.java:270)
     
     
    Error in bfopen (line 114)
    r = bfGetReader(id, stitchFiles);
     
    Error in bfOpen3DVolume (line 50)
    volume = bfopen(filename);
     
    Error in load3DCiliaVolFile (line 4)
    V = bfOpen3DVolume(filepath); 
    ...

and the ImageJ plugin gives an exception on import:

    ImageJ 1.48v; Java 1.6.0_20 [64-bit]; Windows 7 6.1; 20MB of 6075MB (<1%)
    java.lang.NumberFormatException: For input string: "with length indication"
                    at java.lang.NumberFormatException.forInputString(NumberFormatException.java:48)
                    at java.lang.Integer.parseInt(Integer.java:449)
                    at java.lang.Integer.parseInt(Integer.java:499)
                    at loci.formats.in.ZeissCZIReader.initFile(ZeissCZIReader.java:510)
                    at loci.formats.FormatReader.setId(FormatReader.java:1426)
                    at loci.plugins.in.ImportProcess.initializeFile(ImportProcess.java:505)
                    at loci.plugins.in.ImportProcess.execute(ImportProcess.java:143)
                    at loci.plugins.in.Importer.showDialogs(Importer.java:137)
                    at loci.plugins.in.Importer.run(Importer.java:75)
                    at loci.plugins.LociImporter.run(LociImporter.java:78)
                    at ij.IJ.runUserPlugIn(IJ.java:199)
                    at ij.IJ.runPlugIn(IJ.java:163)
                    at ij.Executer.runCommand(Executer.java:131)
                    at ij.Executer.run(Executer.java:64)
 
Oddly, I can't reproduce this behavior locally with the same file, BF version, and MATLAB version (and subsequent Java VM), but I also can't see what else may be causing this problem.  I uploaded the problematic file to https://www.openmicroscopy.org/qa2/qa/upload/ (which I hope went through).  Any help is greatly appreciated!

sbesson, 2015-06-18:
Hi,

thanks for reporting this and uploading a sample file. Using Bio-Formats 5.1.2 I was able to initialize the reader.

Looking at the stack trace, the initialization fails while trying to handle multi-file CZI datasets organized under a folder as:

```
master.czi
master (1).czi
```

I was able to reproduce your stack trace using your sample file and creating an empty file with the same base name ending with ` (with length indication).czi` in the same directory. Could you let us know if such a file would exist or provide us with a list of the files under the directory on the machine where you can reproduce the error? If this is the case, would you expect this CZI file to be part of the same dataset as the original CZI?

Best,
Sebastien

derek-r, 2015-06-19:
Sebastien,

Thanks much for looking into this.  You are correct, there was a file containing the string `(with length indication).czi` in the same directory as the original file.  This was not part of the original dataset but  rather a side-note the data collector made as a reminder as to what was contained within.  I see #1842 looks to correct this issue with behavior I would expect.

Thanks again!

closed 2015-06-19 by derek-r