ome/bioformats #1983: Exporting single plane PNG and TIFF files with bfconvert get corrupted

nylocx, 2015-09-17:
I tested this with the latest release of the Bioformats jar file (5.14 at time of writing)

My goal was to export all timepoints of a lsm file to single RGB16 PNG images.

Steps to reproduce:
1. I used a .lsm file with 1000 timepoints, 3 channels and an XYZ resolution of 512x512x1.
2. Command line for bfconvert was:
```./bfconvert -merge 'input.lsm' 'output_%t.png'```
3. I used pngcheck, imagemagick and graphicsmage to test the resulting pngs (apng images to be precise) which all showed me that they where corrupted.

I then tested it with the .tif extension so I could eventually convert the TIFF images to PNG in a second step but they get corrupted too. Something about an illegal entry for the tiff directory.

I hope that I provided enough information. If you need any more just let me know.

melissalinkert, 2015-09-17:
The PNG issue is in progress here: https://github.com/openmicroscopy/bioformats/pull/1857 (and has been for a while, but we're expecting to have that done for 5.1.5)

nylocx, 2015-09-18:
Oh, nice to hear, so that just leaves the TIFF part that is still not working as expected.

sbesson, 2016-05-03:
Closing this as #1857 is merged. This will be included in Bio-Formats 5.2.0.

closed 2016-05-03 by sbesson