ome/bioformats #2022: ND2 mixup between image index/series in offset calculation?

stevendbrown, 2015-10-13:
When loading a series of images from an ND2 file with `pims` in conjunction with Bioformats, I find that the DeltaT values for a series of planes actually looks like a linear set of acquisition times _across_ series, as if the calculation of the offset in the stream of doubles is being performed as `timestamp_offset = (this_series_number * images_per_series) + this_time_index` rather than `timestamp_offset = (this_time_index * series_count) + this_series_number`. For example, with an ND2 containing 27 FOVs and one image acquired approximate once per minute, I see the following DeltaT values for a single series:
```
[0.9219700636267663,
 1.9037450135946274,
 3.0288364636301996,
 3.95556303858757,
 4.868468838632107,
 5.9362832885980605,
 6.8355563386082645,
 7.827144663631916,
 9.475698738574982,
 10.656013988614083,
 11.653529838621616,
 12.668974513590335,
 13.80917391359806,
 14.803709763586522,
 15.842455963611602,
 16.894236913621427,
 18.006129413604736,
 19.104716288626193,
 20.13068478858471,
 21.167891713619234,
 22.123279888629913,
 23.425835913598537,
 24.38865488862991,
 25.379765988588332,
 26.335722938597204,
 27.349123038589955,
 28.729651288628578,
 60.8162427136302,
 61.46171876358986,
 62.1168248385787,
 62.87410923862457,
 63.655140763580796,
 64.27967733860015,
 64.84931998860836,
 65.45493756359815,
 66.06824346357584,
 66.6852229385972,
 67.31949451363087,
 67.98029468858242,
 68.71559001362324,
 69.374028588593,
 70.02721916359663,
 70.73407038861514,
 71.42024456357956,
 72.31681051361561,
 73.107865688622,
 73.74822493863105,
 74.37522806358338,
 74.97352471357584,
 75.87358053863049,
 76.53868406361342,
 77.42659081357718,
 78.12829901361465,
 79.1507378885746,
 120.96144261360169,
 121.99119918859004,
 123.02214441359042,
 124.00523436361551,
 125.23334946358204,
 126.18084891360998,
 127.14750571358204,
 128.29976698857547,
 129.5479872136116,
 130.77767661362887,
 131.89068833857775,
 133.1268966885805,
 134.13226691359282,
 135.29994918859006,
 136.3480593636036,
 137.45545488858224,
 138.45935091358425,
 139.71330966359378,
 140.73377966362239,
 141.84502403861285,
 142.83436748862266,
 143.79208716362714,
 144.79577266359328,
 145.92105828857422,
 146.87656981360914,
 147.89340736359358,
 149.2373079636097,
 181.21744368863105,
 182.08611658859252,
 182.87250106358528,
 183.5572058135867,
 184.37544556361436,
 185.0481586636305,
 185.94679756361245,
 186.62292186361552,
 187.60025431358815,
 188.35336656361818,
 189.12251141357422,
 189.83873028862476,
 190.55731078863144,
 191.34162231361867,
 192.0975947136283,
 192.86485026359557,
 193.6300066385865,
 194.45286463862658,
 195.28603488862515]
```
I haven't yet figured out how to correct the issue in the Bioformats source code, but if any of the primary developers know what might be happening here, I'd be happy to have some direction towards correcting the issue.

sbesson, 2015-10-13:
Hi @stevendbrown, answering your last question, the timestamp offset logic is performed at https://github.com/openmicroscopy/bioformats/blob/develop/components/formats-gpl/src/loci/formats/in/NativeND2Reader.java#L1367.

As you can see, the offsets are assumed to be ordered by series first and then by plane. This assumption certainly stands for the representative multi-FOV, multi-Z, multi-T nd2 samples we currently have in our data repository as validated via the NIS Element Viewers.

It may be that this assumption on the timestamp offsets does not stand for all nd2 files. To investigate further, would it be possible for you to share a sample nd2 file by uploading it at http://qa.openmicroscopy.org.uk/qa/upload/ if the file is under 2GB? If larger, we can contact you privately and give the credentials to upload them to a FTP server.

stevendbrown, 2015-10-14:
I can acquire an nd2 < 2GB next week and submit it to your upload form.

melissalinkert, 2015-11-24:
Closing, as we haven't seen a file upload; feel free to reopen with a link to the sample file if this problem persists.

closed 2015-11-24 by melissalinkert