ome/bioformats #2060: add support for file format from Olympus FVMPE-RS microscope

garretstuber, 2015-10-31:
Not sure if this is the right place to ask, but the new native file format for Olympus's FVMPE-RS microscope has a new extension (.oir) that is not currently supported by bioformats.  It would be amazingly helpful if we could read in these files.  I would be happy to send files or work with anyone who could help us out with this.  Thanks!

garret

melissalinkert, 2015-11-01:
We do have plans to support this new format; see https://trac.openmicroscopy.org/ome/ticket/13082.

We only have one example file at the moment, so if you are willing to upload one or more files to https://www.openmicroscopy.org/qa2/qa/upload/ that would be greatly appreciated (you may need rename the files or put them in a Zip archive to upload).

melissalinkert, 2015-11-24:
Closing in favor of the above-linked ticket.

closed 2015-11-24 by melissalinkert