ome/bioformats #2079: Problem loading Picoquant .bin files

seanwarren, 2015-11-11:
I'm having some problems loading a Picoquant .bin file using bioformats. The file is here:
https://www.dropbox.com/s/k03jdxgpxhxs9ui/VLDLR_mGFP_with%20ligand%20%20.zip?dl=0

The file is 128x128 pixels with 3125 channels.
I've that it conforms correctly to the format specified by Picoquant: http://www.tcspc.com/doku.php/glossary:pre-histogrammed_image and I'm able to read it correctly in Matlab.

Bioformats however returns a width of 0px and a height of 204,816,160px.

The first 16 bytes of the data are: `80 00 00 00 | 80 00 00 00 | 00 00 20 3f | 35 0c 00 00` which correspond to the width (int32), height(int32), px res (float) and number of channels (int32) respectively. 

The returned height, 204,816,160, is `20 3f 35 0c` in hex. It looks to me like somehow the first 6 bytes might be getting discarded? Looking at PQBinReader.java it's not obvious to me how that might be happening though.

Any ideas would be greatly appreciated! 
@imunro have you seen anything like this before?

Cheers
Sean

imunro, 2015-11-11:
Looking now. If I can fix quickly I might be able to get the fix into the next bf release.

imunro, 2015-11-11:
Hi @seanwarren 

I just imported this to an OMERO server and , although it took forever it looks like the attached. Hope you can see that .
If those dimensions look right then that would suggest it isn’t bio-formats.
Can I ask which version you have?

The slowness is a separate issue related to file size. I know what it is (I can  bore you with details if you want).
If you’re going to be using .bin files of this sort of size with any regularity then we need to address this.
Please let me know.

Ian

[image]

imunro, 2015-11-11:
Just also checked with Ff from source & again I'm getting 128x128x3125.

melissalinkert, 2015-11-24:
@seanwarren, @imunro: is this still an issue with 5.1.6?

imunro, 2015-11-24:
@melissalinkert as in the comments, I failed to reproduce the reported problem with dimensions with bf 5.1.4. & han;t yet tried with 5.1.6 The slowness is a separate issue. If the whole file is below some threshold size (currently 200x200x900), the reader buffers the data in memory and then returns each plane from this buffer. Beyond this size, each plane is read into memory individually (requiring the whole file to be read for each plane) .  Speed-up would require a cleverer buffering system to buffer parts of the file.

sbesson, 2016-04-19:
@imunro: what's the status of this issue with the improvements of https://github.com/openmicroscopy/bioformats/pull/2332?

imunro, 2016-04-19:
It looks, from the comments that bf always got the correct dimensions for this file. 
If the issue was slow loading that should now have been addressed. Would be good to check though but I can't see a QA No for this file.

sbesson, 2016-04-19:
The file mentioned in the PR description has certainly been added to the curated data repository, loading has been tested as part of https://github.com/openmicroscopy/bioformats/pull/2332#issuecomment-208334155 and it is now included as part of the daily Bio-Formats automated tests. Closing this issue. @seanwarren: would it be okay to make this sample file public?

closed 2016-04-19 by sbesson

seanwarren, 2016-04-20:
@sbesson that's fine, but would you mind changing the file name to something more generic please as this is a collaborators data.