ome/bioformats #2110: Invalid pixel values for some MINC 2.0 files?

ctrueden, 2015-11-20:
See [this thread on the ImageJ list](http://imagej.1557.x6.nabble.com/mnc-files-tt5014995.html).

As linked from the thread, sample data illustrating the issue can be [found here](http://www.mouseimaging.ca/mnc/mouse_embryo_atlas/final.model.mnc).

dgault, 2015-11-23:
I was able to reproduce the issue. It appears that the data was being unpacking incorrectly thus leading to noise like values. I have opened a PR to correct the order in which the endianess of the data is set and the pixels unpacked.

https://github.com/openmicroscopy/bioformats/pull/2114

melissalinkert, 2015-11-24:
Closing in favor of https://github.com/openmicroscopy/bioformats/pull/2114.

closed 2015-11-24 by melissalinkert