ome/bioformats #2179: FEITiffReader crashes with some images

avandecreme, 2016-01-13:
With the following image, the FEITiffReader crashes with the following stack trace:
[Tile_001-002-000_0-000.tif.zip](https://github.com/openmicroscopy/bioformats/files/89147/Tile_001-002-000_0-000.tif.zip)

```
java.lang.NumberFormatException: For input string: "1.943794245326921E"
	at sun.misc.FloatingDecimal.readJavaFormatString(FloatingDecimal.java:2043)
	at sun.misc.FloatingDecimal.parseDouble(FloatingDecimal.java:110)
	at java.lang.Double.parseDouble(Double.java:538)
	at java.lang.Double.<init>(Double.java:608)
	at loci.formats.in.FEITiffReader$FEIHandler.characters(FEITiffReader.java:407)
	at org.apache.xerces.parsers.AbstractSAXParser.characters(Unknown Source)
	at org.apache.xerces.impl.XMLDocumentFragmentScannerImpl.scanContent(Unknown Source)
	at org.apache.xerces.impl.XMLDocumentFragmentScannerImpl$FragmentContentDispatcher.dispatch(Unknown Source)
	at org.apache.xerces.impl.XMLDocumentFragmentScannerImpl.scanDocument(Unknown Source)
	at org.apache.xerces.parsers.XML11Configuration.parse(Unknown Source)
	at org.apache.xerces.parsers.XML11Configuration.parse(Unknown Source)
	at org.apache.xerces.parsers.XMLParser.parse(Unknown Source)
	at org.apache.xerces.parsers.AbstractSAXParser.parse(Unknown Source)
	at org.apache.xerces.jaxp.SAXParserImpl$JAXPSAXParser.parse(Unknown Source)
	at org.apache.xerces.jaxp.SAXParserImpl.parse(Unknown Source)
	at javax.xml.parsers.SAXParser.parse(SAXParser.java:195)
	at loci.common.xml.XMLTools.parseXML(XMLTools.java:428)
	at loci.common.xml.XMLTools.parseXML(XMLTools.java:414)
	at loci.common.xml.XMLTools.parseXML(XMLTools.java:393)
	at loci.formats.in.FEITiffReader.initStandardMetadata(FEITiffReader.java:164)
	at loci.formats.in.BaseTiffReader.initMetadata(BaseTiffReader.java:98)
	at loci.formats.in.BaseTiffReader.initFile(BaseTiffReader.java:577)
	at loci.formats.FormatReader.setId(FormatReader.java:1426)
```

This happens because the XML element value "1.943794245326921E-09" get split in two `characters` calls.
According to http://stackoverflow.com/questions/4567636/java-sax-parser-split-calls-to-characters , the processing should be done in `endElement` instead.
I can make a PR to fix that if you want to.

joshmoore, 2016-01-14:
Thanks for the bug report, @avandecreme. A PR would be great if you're willing. We'll include your tif in the test suite to prevent the same from happening again. Cheers.

avandecreme, 2016-01-14:
PR created :)

commit 7e9cd30f (2016-01-20): Merge pull request #2181 from avandecreme/develop

Fix FEITiffReader metadata parser. Fix #2179
commit a7d1a3d4 (2016-01-20): Fix FEITiffReader metadata parser. Fix #2179
commit 47b7b617 (2016-01-25): Merge pull request #2204 from bramalingam/rebased/dev_5_1/develop

Fix FEITiffReader metadata parser. Fix #2179 (rebased onto dev_5_1)
commit 5e16610e (2016-01-25): Fix FEITiffReader metadata parser. Fix #2179
closed 2016-01-20 by sbesson