ome/bioformats #2233: octave package: `bfopen` overrides logging settings and is too verbose

carandraug, 2016-02-08:
The function `bfopen` in the Octave package [overrides the logging settings](https://github.com/openmicroscopy/bioformats/blob/develop/components/formats-gpl/matlab/bfopen.m#L111). The change of logging level should be returned to the previous state. In Octave there is an `unwind_protect` block but I'm not sure how to do that in Matlab.

But is this change of logging level really necessary?

The function is also quite verbose by default, printing to stdout in real time as it reads the images. While some users may prefer this on an interactive session, this is not very nice in other cases.

sbesson, 2016-02-08:
Hi @carandraug 

> The function `bfopen` in the Octave package [overrides the logging settings](https://github.com/openmicroscopy/bioformats/blob/develop/components/formats-gpl/matlab/bfopen.m#L111). The change of [...]

> But is this change of logging level really necessary?

No, it all boils down to the fact that `bfopen` used to be a [standalone utility script](https://github.com/openmicroscopy/bioformats/blob/v4.3.3/components/bio-formats/utils/bfopen.m) for MATLAB. This original script which has been slowly modularized into smaller functions like `bfCheckJavaPath`.

Here, you are right that the logging bit overrides existing preset levels and this behavior could be improved. In the context of u-track, I had started working on a function handling the logging initialization and setup (returning early if not level is passed and the loggers have been initialized). I had not found the time to polish it and port it into the mainline before a major release. If this makes sense and there is a clear driver on the Octave/MATLAB side, I am happy open a PR for discussion.

> The function is also quite verbose by default, printing to stdout in real time as it reads the images. While some users may prefer this on an interactive session, this is not very nice in other cases.

The discussion on the default level can certainly happen in the context of Bio-Formats 5.2.0. Happy to come back to the specifics once I have a PR opened.

carandraug, 2016-02-08:
> Here, you are right that the logging bit overrides existing preset levels and this behavior could be improved. In the context of u-track, I had started working on a function handling the logging [...]

I don't understand. What business does `bfopen` has with setting up logging level? Why not just remove that line completely and use whatever the system is configured to do or whatever the user set before calling `bfopen`?

> If this makes sense and there is a clear driver on the Octave/MATLAB side, I am happy open a PR for discussion.

I can't open a PR without a commit but I don't want to be coding anything before I know what changes will be accepted.

sbesson, 2016-02-08:
>I don't understand. What business does `bfopen` has with setting up logging level?

The minimal specification of the high-level `bfopen` function is to perform the following:

1. check Bio-Formats with `bfCheckJavaPath()`
2. initialize Bio-Formats loggers 
3. initialize a reader using `bfGetReader()` and store the various metadata objects
4. loop over the planes and load the bytes with `bfGetPlane()`

>  Why not just remove that line completely and use whatever the system is configured to do or whatever the user set before calling `bfopen`?

Skipping logging initialization completely means that unless properly configured, at its first call `bfopen()` will outputs `log4j` warnings which are quite user unfriendly:

```
>> bfCheckJavaPath; r=bfGetReader('test.fake');
log4j:WARN No appenders could be found for logger (loci.formats.FormatHandler).
log4j:WARN Please initialize the log4j system properly.
```

That being said, I completely agree that overriding preset levels is not the correct thing to do.

>I can't open a PR without a commit but I don't want to be coding anything before I know what changes will be accepted.

Fair enough. I should have pushed my commit first so that we can have an API discussion - see https://github.com/sbesson/bioformats/commit/be4c93e59eb13b5b1b8ad2b7da0e6e39eaa5d179.

Concretely, the proposal would be to replace step 2 of the workflow above with `bfInitLogging()`. The commit above would preserve the existing `bfopen` functionality *if loggers are not initialized* and set these loggers at `WARN` level by default. If the Bio-Formats loggers have been initialized, the function would simply become a no-op.

closed 2016-03-09 by melissalinkert