ome/bioformats #2237: Exporting to ICS can throw ArrayIndexOutOfBoundsException

ctrueden, 2016-02-10:
Macro to reproduce:
```javascript
run("Organ of Corti (2.8M, 4D stack)");
run("Bio-Formats Exporter", "save=/home/curtis/Desktop/organ-of-corti.ics write_each_channel");
```
Relevant stack trace:
```
(Fiji Is Just) ImageJ 2.0.0-rc-44/1.50e; Java 1.8.0_66 [64-bit]; Linux 3.19.0-39-generic; 55MB of 1065MB (5%)
 
java.lang.ArrayIndexOutOfBoundsException: 15
	at loci.formats.out.ICSWriter.saveBytes(ICSWriter.java:130)
	at loci.formats.FormatWriter.saveBytes(FormatWriter.java:124)
	at loci.plugins.out.Exporter.run(Exporter.java:767)
	at loci.plugins.LociExporter.run(LociExporter.java:75)
```

Exception is the same using both release (Bio-Formats 5.1.7) and dev version (Bio-Formats 5.1.7-DEV 62c4dd810, 9 February 2016).

Reported by @tischi on [this ImageJ forum thread](http://forum.imagej.net/t/bug-in-bio-formats-exporting-ics/823).

bramalingam, 2016-02-11:
Thank you for submitting the issue Curtis. The same was reproducible locally, and a card has been created on this regard. 
https://trello.com/c/L7dtKWsl/96-bio-format-exporter-issues

ctrueden, 2016-02-11:
Thanks @bramalingam !

joshmoore, 2016-02-11:
Closing in favor of the card. Thanks, guys.

closed 2016-02-11 by joshmoore