ome/bioformats #2238: Bio-Formats Exporter can deadlock on OS X

ctrueden, 2016-02-10:
When attempting to reproduce #2237 on OS X, the Bio-Formats Exporter dialog never appears on my OS X 10.11 system. Instead, I get a deadlock:
```
Found one Java-level deadlock:
=============================
"Bio-Formats Exporter":
  waiting to lock monitor 0x000000011a806048 (object 0x00000006444d46d8, a java.lang.Object),
  which is held by "AWT-EventQueue-0"
"AWT-EventQueue-0":
  waiting to lock monitor 0x000000011a804e68 (object 0x000000066aa10ff8, a java.awt.Component$AWTTreeLock),
  which is held by "Bio-Formats Exporter"

Java stack information for the threads listed above:
===================================================
"Bio-Formats Exporter":
	at com.apple.laf.AquaFileSystemModel.getRowCount(AquaFileSystemModel.java:194)
	- waiting to lock <0x00000006444d46d8> (a java.lang.Object)
	at javax.swing.JTable.getRowCount(JTable.java:2662)
	at javax.swing.plaf.basic.BasicTableUI.createTableSize(BasicTableUI.java:1692)
	at javax.swing.plaf.basic.BasicTableUI.getPreferredSize(BasicTableUI.java:1733)
	at javax.swing.JComponent.getPreferredSize(JComponent.java:1661)
	at javax.swing.ScrollPaneLayout.preferredLayoutSize(ScrollPaneLayout.java:495)
	at java.awt.Container.preferredSize(Container.java:1788)
	- locked <0x000000066aa10ff8> (a java.awt.Component$AWTTreeLock)
	at java.awt.Container.getPreferredSize(Container.java:1773)
	at javax.swing.JComponent.getPreferredSize(JComponent.java:1663)
	at java.awt.BorderLayout.preferredLayoutSize(BorderLayout.java:719)
	- locked <0x000000066aa10ff8> (a java.awt.Component$AWTTreeLock)
	at java.awt.Container.preferredSize(Container.java:1788)
	- locked <0x000000066aa10ff8> (a java.awt.Component$AWTTreeLock)
	at java.awt.Container.getPreferredSize(Container.java:1773)
	at javax.swing.JComponent.getPreferredSize(JComponent.java:1663)
	at java.awt.BorderLayout.preferredLayoutSize(BorderLayout.java:719)
	- locked <0x000000066aa10ff8> (a java.awt.Component$AWTTreeLock)
	at java.awt.Container.preferredSize(Container.java:1788)
	- locked <0x000000066aa10ff8> (a java.awt.Component$AWTTreeLock)
	at java.awt.Container.getPreferredSize(Container.java:1773)
	at javax.swing.JComponent.getPreferredSize(JComponent.java:1663)
	at javax.swing.BoxLayout.checkRequests(BoxLayout.java:483)
	at javax.swing.BoxLayout.layoutContainer(BoxLayout.java:424)
	- locked <0x0000000644741298> (a javax.swing.BoxLayout)
	at java.awt.Container.layout(Container.java:1503)
	at java.awt.Container.doLayout(Container.java:1492)
	at java.awt.Container.validateTree(Container.java:1688)
	at java.awt.Container.validateTree(Container.java:1697)
	at java.awt.Container.validateTree(Container.java:1697)
	at java.awt.Container.validateTree(Container.java:1697)
	at java.awt.Container.validateTree(Container.java:1697)
	at java.awt.Container.validate(Container.java:1623)
	- locked <0x000000066aa10ff8> (a java.awt.Component$AWTTreeLock)
	at java.awt.Container.validateUnconditionally(Container.java:1660)
	- locked <0x000000066aa10ff8> (a java.awt.Component$AWTTreeLock)
	at java.awt.Window.pack(Window.java:818)
	at javax.swing.JFileChooser.createDialog(JFileChooser.java:805)
	at javax.swing.JFileChooser.showDialog(JFileChooser.java:732)
	at javax.swing.JFileChooser.showSaveDialog(JFileChooser.java:664)
	at loci.plugins.out.Exporter.run(Exporter.java:236)
	at loci.plugins.LociExporter.run(LociExporter.java:75)
	at ij.plugin.filter.PlugInFilterRunner.processOneImage(PlugInFilterRunner.java:263)
	at ij.plugin.filter.PlugInFilterRunner.<init>(PlugInFilterRunner.java:112)
	at ij.IJ.runUserPlugIn(IJ.java:214)
	at ij.IJ.runPlugIn(IJ.java:176)
	at ij.Executer.runCommand(Executer.java:136)
	at ij.Executer.run(Executer.java:65)
	at java.lang.Thread.run(Thread.java:745)
"AWT-EventQueue-0":
	at java.awt.Component.invalidate(Component.java:2920)
	- waiting to lock <0x000000066aa10ff8> (a java.awt.Component$AWTTreeLock)
	at java.awt.Container.invalidate(Container.java:1580)
	at javax.swing.JComponent.revalidate(JComponent.java:4862)
	at javax.swing.JTable.tableRowsInserted(JTable.java:4482)
	at javax.swing.JTable.tableChanged(JTable.java:4407)
	at javax.swing.table.AbstractTableModel.fireTableChanged(AbstractTableModel.java:296)
	at javax.swing.table.AbstractTableModel.fireTableRowsInserted(AbstractTableModel.java:231)
	at com.apple.laf.AquaFileSystemModel$DoChangeContents.run(AquaFileSystemModel.java:458)
	- locked <0x00000006444d46d8> (a java.lang.Object)
	- locked <0x000000063fc952f0> (a java.lang.Object)
	at java.awt.event.InvocationEvent.dispatch(InvocationEvent.java:312)
	at java.awt.EventQueue.dispatchEventImpl(EventQueue.java:745)
	at java.awt.EventQueue.access$300(EventQueue.java:103)
	at java.awt.EventQueue$3.run(EventQueue.java:706)
	at java.awt.EventQueue$3.run(EventQueue.java:704)
	at java.security.AccessController.doPrivileged(Native Method)
	at java.security.ProtectionDomain$1.doIntersectionPrivilege(ProtectionDomain.java:76)
	at java.awt.EventQueue.dispatchEvent(EventQueue.java:715)
	at java.awt.EventDispatchThread.pumpOneEventForFilters(EventDispatchThread.java:242)
	at java.awt.EventDispatchThread.pumpEventsForFilter(EventDispatchThread.java:161)
	at java.awt.EventDispatchThread.pumpEventsForHierarchy(EventDispatchThread.java:150)
	at java.awt.EventDispatchThread.pumpEvents(EventDispatchThread.java:146)
	at java.awt.EventDispatchThread.pumpEvents(EventDispatchThread.java:138)
	at java.awt.EventDispatchThread.run(EventDispatchThread.java:91)
```

Happens with Oracle Java 1.7.0_80 as well as Oracle Java 1.8.0_66. And it happens with both the release and 5.1 dev versions of Bio-Formats.

ctrueden, 2016-02-10:
It seems odd that `FileChooser#showSaveDialog` deadlocks when called off the EDT on OS X, but I guess the fix could be to wrap that call in `EventQueue.invokeAndWait`...

bramalingam, 2016-02-11:
Thank you for submitting the issue Curtis. The same was reproducible locally, and a card has been created on this regard. 
https://trello.com/c/L7dtKWsl/96-bio-format-exporter-issues

ctrueden, 2016-02-11:
Thanks @bramalingam !

joshmoore, 2016-02-11:
Closing in favor of the card. Thanks, guys.

closed 2016-02-11 by joshmoore

dgault, 2022-01-07:
Further reports of this issue of imagesc - https://forum.image.sc/t/problem-with-bio-format-exporter/58192/2
The post includes some sceanrios for reproducing the issue and details of the user system

dgault, 2022-05-19:
Another report of this, also on MacOS Mojave: https://forum.image.sc/t/export-image-sequence-as-avi-and-bio-formats/67199