ome/bioformats #2255: NIOFileHandle large buffer size detracts from ImageJ importer performance

etarena, 2016-02-19:
We noticed that when opening OME-TIFF datasets produced by WiscScan, the initialization step took a long time. Profiling the code, we found that much time was spent in `NIOByteBufferProvider` in the `allocateDirect` method, copying data from the `FileChannel` into the buffers. The default read-only buffer size is set to 1048576 bytes, which gets allocated for every TIFF file in the dataset, causing the `channel.read` call to take a long time populating each buffer.

The issue can be significantly mitigated by calling `NIOFileHandler.setDefaultBufferSize(int)` with a smaller value. Here is a quick benchmark of the difference with various buffer sizes, when reading an (fill in size of dataset) OME-TIFF dataset from an SMB networked file system:
```
Buffer size: 16 Process time: 90566
Buffer size: 1024 Process time: 92513
Buffer size: 8192 Process time: 96178
Buffer size: 65536 Process time: 111689
Buffer size: 1048576 Process time: 313106
```
Here is the script which generated the benchmark:
```javascript
var list = newArray(16, 1024, 8192, 65536, 1048576);
for (i = 0; i < list.length; i++) {
	call("loci.common.NIOFileHandle.setDefaultBufferSize",list[i]);
	start = getTime();
	run("Bio-Formats", "open=[/Volumes/data/Jayne/test data sets for Ellen/3ROI 2filters 2 channels 3TP/H150901R1 012016 postW_TL1 2dpfDish1 krtGFP_SHG 20x890nmp50_160_z2_5i2g580f520_445u5ROI3TP6_C0_TP0_SP0_FW0.ome.tiff] autoscale color_mode=Default open_all_series view=Hyperstack stack_order=XYCZT");
	end = getTime();
	print("Buffer size: " + list[i] + " Process time: " + (end - start));
	run("Close All");
}
```
You can find the data at the above path on LOCI ftp accessible via the usual credentials (just message a LOCI programmer if you need them). 

As you can see above, the smaller the buffer size the faster the data is opened within ImageJ.  We could certainly hardcode the buffer size within ImageJ - but we are wondering if you have another idea? 

Many thanks for your insight.

melissalinkert, 2016-02-22:
Thank you for reporting this.  I don't see the dataset after logging in to dev.loci.wisc.edu (the last ftp server known to me) - could you please upload the files to https://www.openmicroscopy.org/qa2/qa/upload/ instead?

etarena, 2016-02-23:
@melissalinkert - the dataset has been uploaded.  (note: some of the images are 'incomplete' - datasets for some time points is missing as the acquisition was ended early in this case...)

melissalinkert, 2016-02-25:
Thanks, @etarena - I can confirm the results above with the files in QA 17039.  I'd be a little hesitant to recommend hard-coding a smaller buffer size without further investigation into how that will affect the performance of other formats, though it's probably safe as a temporary measure if you're only working with OME-TIFFs.

The next step from our side is to try a few of the smaller sizes against a larger set of our test data across several formats - if that has a noticeable positive effect, then we may lower the default buffer size across the board.  Otherwise, the easiest thing is will be to change all ```RandomAccessInputStream``` instantiations in ```OMETiffReader``` to look like https://github.com/openmicroscopy/bioformats/blob/develop/components/formats-bsd/src/loci/formats/in/OMETiffReader.java#L125.  It will take us a couple of days to finish properly investigating though.

melissalinkert, 2016-04-11:
Pull request opened now that we've had a chance to do some more thorough checking with the smaller buffer size: https://github.com/openmicroscopy/bioformats/pull/2338

Closing this issue; can certainly continue the discussion on the pull request as needed.

closed 2016-04-11 by melissalinkert