ome/bioformats #2361: CZI: autoselect series for file from multi-file dataset

stelfrich, 2016-04-20:
When opening one file from a multi-file dataset with Bioformats from ImageJ, the series selection window shows the available series and selects the first series by default.

Would it be possible to make the series of the opened file the default selection?

rleigh-dundee, 2016-04-22:
Please see https://github.com/openmicroscopy/bioformats/issues/2309#issuecomment-206392411

The problem is that there isn't a 1:1 correspondence between "file" and "series".  A file might contain multiple series, one series, or a part of a series split over multiple files.  This depends entirely upon the file format and acquisition parameters.  For example, an OME-TIFF multi-file set could support all of these layouts.  So making this default-to-series-in-opened-file behaviour isn't something which we can solve for the general case, unfortunately.

sebi06, 2016-04-25:
Hi guys,

which way was this mCZI file created? Is it a multi-scene image from ZEN Blue or Black acquired with "Spit Scenes into Separate Files" option activated?

stelfrich, 2016-04-26:
Thanks for the pointer to the previous discussion, @rleigh-dundee!

> which way was this mCZI file created? Is it a multi-scene image from ZEN Blue or Black acquired with "Spit Scenes into Separate Files" option activated?

Indeed, @sebi06. Files were generated with an older ZEN Blue.

closed 2016-04-26 by sbesson