ome/bioformats #2367: New warining message about SlideBook6Reader.dll with 5.1.9

sebi06, 2016-04-25:
Hi guys,

when i use the bioformats_package.jar from 5.1.9 from python I get those warnings. But when I switch back to version 5.1.8 those warnings do not pop up. I both cases my CZI file is read correctly, but I am just wondering...
When I open the file directly from within Fiji no warning appears no matter what version (5.1.8 or 5.1.9) no  warning appears. 

Sebi 

C:\Anaconda\python.exe C:/Users/M1SRH/Documents/Python_Projects/BioFormatsRead/test_get_image6d.py
09:36:35.054 [Thread-0] DEBUG loci.common.NIOByteBufferProvider - Using mapped byte buffer? false
Apr 25, 2016 9:36:35 AM org.scijava.nativelib.BaseJniExtractor extractJni
INFORMATION: Couldn't find resource META-INF/lib/windows_64/ SlideBook6Reader.dll
Apr 25, 2016 9:36:35 AM org.scijava.nativelib.NativeLibraryUtil loadNativeLibrary
WARNUNG: IOException creating DefaultJniExtractor
java.io.IOException: Couldn't find resource META-INF/lib/windows_64/ SlideBook6Reader.dll
	at org.scijava.nativelib.BaseJniExtractor.extractJni(BaseJniExtractor.java:144)
	at org.scijava.nativelib.NativeLibraryUtil.loadNativeLibrary(NativeLibraryUtil.java:265)
	at loci.formats.in.SlideBook6Reader.<clinit>(SlideBook6Reader.java:93)
	at java.lang.Class.forName0(Native Method)
	at java.lang.Class.forName(Unknown Source)
	at loci.formats.ClassList.<init>(ClassList.java:127)

bramalingam, 2016-04-27:
Hi,

Thank you for reporting your issue.

Bio-Formats reads a file with each reader detect the format. In our recent 5.1.9 release of Bio-Formats, we removed the native binary files required for the Slidebook reader from our repository (https://github.com/openmicroscopy/bioformats/pull/2300) which led to the warnings for the “missing Slidebook dll’s” in your script.

I have created a ticket on this regard, https://trac.openmicroscopy.org/ome/ticket/13220, and we are currently working on a fix. We are looking at releasing a patched version at the earliest.

Hope that helps.

Best,
Balaji

sbesson, 2016-04-27:
Closing in favor of the ticket.

closed 2016-04-27 by sbesson

sbesson, 2016-05-05:
See also http://forum.imagej.net/t/closing-console-in-fiji/1531