ome/bioformats #2370: Image Dimension problem with CZIReader

sebi06, 2016-04-25:
Hi guys,

I just realized a serious problem with the CZIReader of BioFormats which puszzles me. Here is what I acquired in ZEN:

4 Scenes with a 2x2 TileRegion per scene --> 16 image series
3 Timepoints
5 Z-Planes
2 Channels
Camera ROI = 640x640
Overlap = 10%

When I open this one in Fiji the XML-MetaInfo show an image size of 1216x1216, which the the size of on 2x2 TileRegion with 10% overlap using a 640x640 ROI on the camera. But the result in Fiji is strange, since all the single tile are always placed wrong.

I expected to see an image size of 640 x 640 and a stack of 16 image series. See the attached screenshot.

But when I adress the single image series using python_bioformats, the size of a single tile is shown correctly as 640x640.

I uploaded the CZI to my dropbox account: https://dl.dropboxusercontent.com/u/623476/20160425_BF_CZI.czi

Any help or clarification is greatly appriciated, since those black empty areas are not really usefull. The CZI was acquired with the lates upcoming ZEN Blue 2.3 Release, which is about to be annouced officially.

Cheers,

Sebi (from Zeiss)

[image]

melissalinkert, 2016-04-26:
Thanks, @sebi06.  I can confirm with 5.1.9 and latest develop snapshot.  We'll either need to adjust the SizeX/SizeY to be 640x640 or adjust the prestitching settings so that only 4 series are reported.

Added to https://trac.openmicroscopy.org/ome/ticket/12935 for consideration with other 5.2.0-related .czi work; if you would like to be CC'd on that ticket, please let us know.

closed 2016-04-26 by melissalinkert

sebi06, 2016-04-27:
Hi Melissa,

as far as I know, the CZIReader within BioFromats always interpreted single tiles withan a CZI as image series. TileRegion were never supported since the OME-TIFFformat does not have a scene concept.

So what is this pre-stitching setting all about? Does this work for any arbitrary tile region, because ZEN acquires all kind of tile region, not only rectangular ones. This would mean, that the CZIReader would have to place every single tile according the its XYZ coordinates within the image canvas to be displayed inside Fiji & Co, right? This sounds tempting but quite challenging to me.

And what puzzels me as well, that the problem mentioned above is new. As far as I remember a x*y tile CZI was always display as x*y image series stack, where the canvas had the size of one single tile.

Furthermore, we I use BioFormats (5.1.9) via Python, the same file will be read differently without the problems mentioned above. Here is my console output:

Image Directory            :  c:\Users\M1SRH\Documents\Python_Projects\BioFormatsRead
Image Filename           :  20160425_BF_CZI.czi
Images Dim Sizes        :  [16, 3, 5, 2, 640, 640]
Dimension Order*        :  XYCZT
Dimension Order CZI   :  BSTCZYX0
Total Series Number     :  16
Image Dimensions       :  16 3 5 2 640 640
Scaling XYZ [micron]   :  0.2 0.2 0.5
Objective M-NA-Imm   :  5.0 0.35 Air
Objective Name           :  Plan-Apochromat 5x/0.35
Ex. Wavelengths [nm]  :  [ 493.  553.]
Em. Wavelengths [nm] :  [ 517.  568.]
Dyes                            :  ['None', 'None']
Detector Model            :  TestCam
Detector Name            :  Detector:Internal
Channels                     :  ['AF488', 'AF555']
Channel Description    :  []
Array Shape 6D           :  (16L, 3L, 5L, 2L, 640L, 640L)

I use a line like this:

`MetaInfo['SizeY'] = np.int(jmd.getPixelsSizeY(IMAGEID).getValue().floatValue())`

and with the IMAGEID=0 this yields in 640 for the test data. And when I try this  differently, I also get 640 x 640 as size.

`# get dimensions for CTZXY`
`md = bioformats.OMEXML(new_omexml)`
`pixels = md.image(IMAGEID).Pixels`
`SizeX = pixels.SizeX`
`SizeY = pixels.SizeY`
  
Any idea, why this happens?

sebi06, 2016-04-27:
Hi Melissa,

i still habe various question regarding this issue, but I cannot reopen it.
So I added my comments to this closed issue, which was probably not a good
idea. So I also added a new issue (i Hope this was ok).

Furthermore, do you have the latetes official CZI File Format
Documentation. I a few days the new ZEN 2.3 version will be released, so it
might be useful to check?

Please let me know if you need more testdata or anything else.

Sebi