ome/bioformats #2377: Image Dimension Issue #2370 - Related issues when using BioFormats from Python

sebi06, 2016-04-27:
This related to issue #2370 .

Testfile: https://dl.dropboxusercontent.com/u/623476/20160425_BF_CZI.czi

[image]

As far as I know, the CZIReader within BioFromats always interpreted single tiles within a CZI as image series. TileRegion were never supported since the OME-TIFF format does not have a scene concept.

So what is this pre-stitching setting all about? Does this work for any arbitrary tile region, because ZEN acquires all kind of tile region, not only rectangular ones. This would mean, that the CZIReader would have to place every single tile according the its XYZ coordinates within the image canvas to be displayed inside Fiji & Co, right? This sounds tempting but quite challenging to me.

And what puzzels me as well, that the problem mentioned above is new. As far as I remember a x*y tile CZI was always display as x*y image series stack, where the canvas had the size of one single tile.

Furthermore, we I use BioFormats (5.1.9) via Python, the same file will be read differently without the problems mentioned above. Here is my console output:

Image Directory            :  c:\Users\M1SRH\Documents\Python_Projects\BioFormatsRead
Image Filename           :  20160425_BF_CZI.czi
Images Dim Sizes        :  [16, 3, 5, 2, 640, 640]
Dimension Order*        :  XYCZT
Dimension Order CZI   :  BSTCZYX0
Total Series Number     :  16
Image Dimensions       :  16 3 5 2 640 640
Scaling XYZ [micron]   :  0.2 0.2 0.5
Objective M-NA-Imm   :  5.0 0.35 Air
Objective Name           :  Plan-Apochromat 5x/0.35
Ex. Wavelengths [nm]  :  [ 493.  553.]
Em. Wavelengths [nm] :  [ 517.  568.]
Dyes                            :  ['None', 'None']
Detector Model            :  TestCam
Detector Name            :  Detector:Internal
Channels                     :  ['AF488', 'AF555']
Channel Description    :  []
Array Shape 6D           :  (16L, 3L, 5L, 2L, 640L, 640L)

I use a line like this:

`MetaInfo['SizeY'] = np.int(jmd.getPixelsSizeY(IMAGEID).getValue().floatValue())`

and with the IMAGEID=0 this yields in 640 for the test data. And when I try this  differently, I also get 640 x 640 as size.

`# get dimensions for CTZXY`
`md = bioformats.OMEXML(new_omexml)`
`pixels = md.image(IMAGEID).Pixels`
`SizeX = pixels.SizeX`
`SizeY = pixels.SizeY`
  
Any idea, why this happens?

sebi06, 2016-05-11:
I digged a bit deeper within our ZEN software to find the reason for the issue #2370 I recently reported on GitHub. I tried to summarize some info inside the attached PDF.

The short version  is: The order of sublocks inside a CZI is not sufficient to decide if an individual sublock is a “real” image subblock or just a “pyramid sublock”. This has to do with optimizations inside ZEN during saving the image data. To me it looks like BioFormats is just looking for the size of the first sublock, which may or may not be a “real” subblock. Does this makes sense to you?
[Pyramid_Subblock_Issue_BioFormats.pdf](https://github.com/openmicroscopy/bioformats/files/258734/Pyramid_Subblock_Issue_BioFormats.pdf)

sebi06, 2016-06-01:
Hi guys,

any news on this issue? Do you need more information or help? Did someone read the attached PDF?

Sebi

hflynn, 2016-06-02:
Hi @sebi06 apologies for no-one having got back to you yet. We are all really busy with our annual user meeting this week but I'll make sure someone looks at this asap next week.

melissalinkert, 2016-06-08:
Thanks for writing this up, @sebi06, and our apologies again for not getting back to you sooner.  Yes, this all makes sense, and I can confirm locally the behavior noted in the above screenshot (with 5.1.10) and how this differs from what is shown in Zen.  5.1.9 and 5.1.10 show the same dimensions locally; I have not so far been able to find a 5.1.x or 5.0.x version that reports the 640x640 dimensions reported above.  I assume you are using https://github.com/sebi06/BioFormatsRead, so I do wonder if somehow czifile.py is being used instead of Bio-Formats (though I admit to not understanding the full details of how that package works).

In any case, we are aware of various inconsistencies with how dimensions are detected and reported when multiple tiles and/or scenes are present; this depends (to our knowledge) on differences in the acquisition settings and post-processing steps that were performed.  We expect to make substantial progress on this in the coming weeks, in preparation for 5.2.0.

I've linked this report to an existing ticket: https://trac.openmicroscopy.org/ome/ticket/13233

...and we'll post updates there as appropriate.  We can certainly CC you on the ticket if you like; any resulting fix will be automatically cross-linked both to the ticket and this issue.

sebi06, 2016-06-09:
Hi Melissa,

thank for getting back. It all comes down to the fact, that the order of
image sublocks in a CZI file, depends on the way the image data where
acquire. So in some cases the first blocks are already "real" image blocks
(Scale = 1). But in other cases it might by pyramid blocks, where the scale
is different.

I know this is nasty from a BioFormats point of view, you need to check,
what type (or scale) a block has, before using its dimenions. When assuming
a pyramid block for a real image block you get exactly the issues decribed.
Let me know if you need any help with more detail info about the CZI and
its structure. The guys programming the file format are just next door ...

Cheers,

Sebi (From Zeiss)

melissalinkert, 2018-12-21:
Just tested this file with 5.9.2, and it seems to be working correctly.  8 series are reported, 2 per scene; the first series for each scene is 1216x1216, the second is 608x608 and both are identified as a single image pyramid.  Opening all planes in ImageJ results in images that match what I see in ZEN (i.e. no empty space as in the screenshot above).

Closing as there doesn't seem to be anything else to fix here, but feel free to reopen if you notice the same problem with 5.9.2.

closed 2018-12-21 by melissalinkert