ome/bioformats #2400: Andor iQ TIFF metadata issues

JimboMahoney, 2016-05-05:
Hi OME team!

I've found two problems with the metadata read from Andor iQ's TIFFs. I'm using the latest version of Andor iQ (3.4.1) and Bio-Formats (5.1.9, 14th April 2016):

1) Z resolution defaults to 1. e.g. see the following from the metadata:

Pixels BigEndian="false" DimensionOrder="XYCZT" ID="Pixels:0" Interleaved="false" PhysicalSizeX="0.116669359036491" PhysicalSizeXUnit="µm" PhysicalSizeY="0.116669359036491" PhysicalSizeYUnit="µm" **PhysicalSizeZ="1.0"** PhysicalSizeZUnit="µm" SignificantBits="16" SizeC="2" SizeT="20" SizeX="320" SizeY="240" SizeZ="21" TimeIncrement="1.0" TimeIncrementUnit="s" Type="uint16"

In the dataset I've uploaded to you (Protocol114.tif), the Z-step is actually 0.5 um.

2) Z position is out by a factor of 1000. e.g. see the following from the metadata:

Plane DeltaT="0.0" DeltaTUnit="s" PositionX="0.0" PositionXUnit="reference frame" PositionY="0.0" PositionYUnit="reference frame" **PositionZ="-0.0049962**" PositionZUnit="µm" TheC="0" TheT="0" TheZ="0"/

This should be "-4.9962"

I've uploaded a dataset (Protocol114.tif) and the .txt file of its metadata here:

http://qa.openmicroscopy.org.uk/qa/upload/

Let me know if you need anything else?

JimboMahoney, 2016-05-05:
Dataset here:

https://www.openmicroscopy.org/qa2/qa/feedback/17175/

sbesson, 2016-05-10:
Closing in favor of the reference Trac ticket - https://trac.openmicroscopy.org/ome/ticket/13123.

closed 2016-05-10 by sbesson