ome/bioformats #2431: ZeissCZIReader throws a NumberFormatException

gatagat, 2016-05-25:
Hi all,

we have a couple of files that make the ``ZeissCZIReader`` to fail (using bioformats from within imagej).

Trying to check the file with showinf results in the following stacktrace:

    Exception in thread "main" java.lang.NumberFormatException: null
        at java.lang.Integer.parseInt(Integer.java:443)
        at java.lang.Integer.parseInt(Integer.java:514)
        at loci.formats.in.ZeissCZIReader.translateExperiment(ZeissCZIReader.java:2268)
        at loci.formats.in.ZeissCZIReader.translateMetadata(ZeissCZIReader.java:1329)
        at loci.formats.in.ZeissCZIReader.initFile(ZeissCZIReader.java:731)
        at loci.formats.FormatReader.setId(FormatReader.java:1426)
        at loci.formats.ImageReader.setId(ImageReader.java:835)
        at loci.formats.ReaderWrapper.setId(ReaderWrapper.java:651)
        at loci.formats.tools.ImageInfo.testRead(ImageInfo.java:1028)
        at loci.formats.tools.ImageInfo.main(ImageInfo.java:1117)

The bioformats version should be up to date:

    ./showinf -version
    Version: 5.1.10
    VCS revision: 3125500841deff07687b25eb3f430dca62dee648
    Build date: 6 May 2016

Let me know where can I upload the test data.

Thanks for any help.

Best regards,
Tomas

gatagat, 2016-05-25:
A similar issue has been discussed on the ome-users mailing list:
http://lists.openmicroscopy.org.uk/pipermail/ome-users/2015-January/004988.html
From the discussion there it seems it was resolved and a patch merged, however I see the error with the current version (5.1.10).

Another mention of the issue just two months ago on the imagej mailing list:
http://permalink.gmane.org/gmane.comp.java.imagej/40304

dgault, 2016-05-25:
Hi Tomas,

Thank you for reporting this issue. There have been similar issues with NumberFormatExceptions reported and fixed in the ZeissCZIReader previously, however looking at the code there may still be further checks that need put in place for certain metadata combinations.

If you could upload a sample file I will be able to reproduce and further investigate the issue - https://www.openmicroscopy.org/qa2/qa/upload/ 

With Thanks,
David Gault

dgault, 2016-05-26:
Hi Tomas,

Thank for providing a QA file in order to help reproduce this issue.
I have put in place a fix and opened a PR to be tested and integrated into a future Bio-Formats release.

With Thanks,
David Gault

gatagat, 2016-05-26:
Hi David,

thanks for the superfast fix! Any idea when this will be out for normal users (~update in fiji)? Just curious.

Best regards,
Tomas

dgault, 2016-06-01:
Hi Tomas,

Apologies for the delayed response, I was on holiday at the end of last week and we are holding our annual user meeting this week. I don't have an estimated date at the moment, once the PR is reviewed, tested and approved then it will later need to be scheduled into a future release.

With Thanks,
David Gault

sbesson, 2016-06-08:
See also the [ome-users thread](http://lists.openmicroscopy.org.uk/pipermail/ome-users/2016-June/006056.html) reporting the same issue. Closing this in favor of the opened Pull Request.

closed 2016-06-08 by sbesson