ome/bioformats #2467: Bug in CZI Lightsheet Reader

StephanPreibisch, 2016-06-22:
Hi,

I realised a pretty significant bug in the Zeiss Lightsheet CZI reader. It does read the sizes of the individual stacks wrongly (assumes the biggest stack size for all of them), which is a huge problem as we need to correct it by hand for every acquisition, otherwise post-processing fails:

'Information|Image|V|View|SizeZ #1': 78

WRONG, this view really has 73 slices
'Information|Image|V|View|SizeZ #2': 78

CORRECT, this view has 78 slices
'Information|Image|V|View|SizeZ #3': 78

WRONG, this view really has 77 slices
'Information|Image|V|View|SizeZ #4': 78

WRONG, this view really has 54 slices
It just shows the maximum number of slices of all views, not the correct amount per view. I uploaded the corresponding dataset here: http://preibischlab.mdc-berlin.de/download/dros-3.czi (1.7GB)

It has 4 angles (0,90,180,270), two channels (488 & 561), one illumination direction and one timepoint.

Please let me know if I can do anything else to help fix this.

All the best from Berlin,
Stephan

StephanPreibisch, 2016-07-08:
Hi, I forgot to mention, that this did work before and broke at some point. This is really quite a big problem for analysing the data, do you have any idea when you have time to look into it?

If it is not possible in the near future I would start implementing a work-around to detect zero planes in the opened stacks to correct. Please let me know. Thanks a lot, Stephan

dgault, 2016-07-11:
Hi Stephan,

First off my apologies for the long delay in responding to this issue and thank you for making available a sample file to test with.

I will begin investigating this today to see what possible solutions we have and follow up with you with my findings.

With Thanks,
David Gault

StephanPreibisch, 2016-07-11:
Great, thanks a lot!

dgault, 2016-07-13:
Hi Stephan,

Just as a follow up to this issue, I have been working on a solution which appears to resolve these issues. This will have to go through further testing and a PR opened and reviewed before it can be scheduled for any future releases. Once I have the Pull Request opened I shall let you know so you can keep track of the fix.

With Thanks,
David Gault

StephanPreibisch, 2016-07-13:
Thanks so much David, we really need this fix urgently. If I can help somehow please let me know.

All the best, Stephan