ome/bioformats #2485: MicroManagerReader - Incorrect PixelType

dgault, 2016-07-11:
A number of issues were discovered around the parsing of PixelType values after the submission of sample data from a user. Firstly when parsing the metadata.txt file and reading the Key 'IJType' we currently support the following PixelTypes:
0 - UINT8
1 - UINT16

In this particular case we are seeing IJType with a value of 4. As such we need to set appropriate values for the remaining values in the 'IJType' range.

Secondly after parsing the metdata.txt file and setting the PixelType, we are then parsing the Tiff files. During this process we read the PixelType from the first tiff file (in this case giving us an incorrect value of UINT8) and overwriting the previously set value from the metadata.txt file.

@marktsuchida @ChrisWeisiger Would you be able to help with the IJType values we are missing and provide advice on dealing with reading PixelType from the tiff file?

ChrisWeisiger, 2016-07-11:
Hello, yes, we're happy to help on this issue.

The IJType values are taken from ImageJ (specifically a set of constants in `ij.ImagePlus`). A currently-exhaustive list of those values:
* 0: 8-bit grayscale
* 1: 16-bit grayscale
* 2: 32-bit floating point grayscale
* 3: 8-bit indexed color
* 4: 32-bit RGB color (8 bits per component)

In practice we do not use modes 2 or 3 currently. We have no plans to support mode 3, and while we would like to support mode 2 sometime, we have no concrete plans to do so. So really all that's needed here is to add support for the RGB color mode.

Regarding the per-image PixelType, in my tests in MM2.0 with demo hardware, if I save an RGB image and then examine the metadata, I get a consistent PixelType of "RGB32" between the summary JSON and the per-image JSON. I've uploaded this test file as rgbtest_1.tgz (and feel free to make that file public if you want; there's nothing special in it). Could you please clarify precisely which fields you're reading and the values you're getting from them?

dgault, 2016-07-11:
Hi Chris,

I tested the rgbtest_1.tgz you uploaded and am seeing the same results with it. When reading the Pixel type from the tiff we are basing that value on a combination of BitsPerSample (tag 258) and SampleFormat (tag 339). So for the rgbtest it is reading BitsPerSample as 8 and SampleFormat as -1.

ChrisWeisiger, 2016-07-11:
Okay, this is getting into parts of the code that I'm not an expert on, and am correspondingly reluctant to modify without a very good idea of what the implications are. But if I'm reading our saving code correctly, when we have an RGB image, we allocate 6 bytes for the BitsPerSample tag (I'd guess 2 bytes per component in the image). Presumably the intended interpretation is that you read three different components (because the tag indicates there are 3 items in it) and sum them together to get 24 bits per sample.

I don't see tag 339 being written anywhere in our code, which is probably why it's reading as -1. Is there a different value that should be written for a correct interpretation of multi-component images?

dgault, 2016-07-12:
Ive done some further testing, using the BitsPerSample along with SamplesPerPixel tag. This gives me some progress but even accounting for the SampleFormat tag not being set it would seem that there are still further issues in our MicroManagerReader, largely around the reading of the RGB samples. I will continue investigating to see if a fix can be put in place at our reader level.