ome/bioformats #2490: Add zero padding to bfconvert pattern to fix sorting

burgerga, 2016-07-20:
When I use `bfconvert` on a large XYZCT stack with 

    bfconvert  sample.nd2 output_xy%sz%zc%ct%t.ome.tif

I get the following output files:

```
output_xy0z0c0t0.ome.tif
output_xy0z0c0t10.ome.tif
output_xy0z0c0t11.ome.tif
output_xy0z0c0t12.ome.tif
output_xy0z0c0t13.ome.tif
output_xy0z0c0t14.ome.tif
output_xy0z0c0t15.ome.tif
output_xy0z0c0t16.ome.tif
output_xy0z0c0t17.ome.tif
output_xy0z0c0t18.ome.tif
output_xy0z0c0t19.ome.tif
output_xy0z0c0t1.ome.tif
output_xy0z0c0t20.ome.tif
output_xy0z0c0t21.ome.tif
output_xy0z0c0t22.ome.tif
output_xy0z0c0t23.ome.tif
output_xy0z0c0t24.ome.tif
output_xy0z0c0t25.ome.tif
output_xy0z0c0t26.ome.tif
output_xy0z0c0t27.ome.tif
```

Would it be possible to add zero padding (in a printf like way) to get natural sorting?

dgault, 2016-07-21:
Hi Gerhard,

There are 2 possible ways we could do this:
- Automatically add the zero padding based on the max size of the parameters
- Add a new option to the API to allow for formatting (like printf)

I have put together some initial code which would enable the first option - https://github.com/dgault/bioformats/commit/1ac64b36629d92d15656d2837a3ea237b77b9165

As this has the potential for an API change there will have to be further discussion on the best possible solution. If you have any preferences on which you would like to see it would be good to know.

With Thanks,
David

burgerga, 2016-07-21:
@dgault Hi David,

Thanks for your fast response!
I think I prefer the first option, because it's simple to use (and to implement ;) ). At them moment I can't think of use cases where option 2 would be worth the effort.

sbesson, 2016-09-13:
Should be fixed in [Bio-Formats 5.2.2](https://www.openmicroscopy.org/community/viewtopic.php?f=11&t=8118).

closed 2016-09-13 by sbesson