ome/bioformats #2540: [Linux] Make bftools symlink-able

burgerga, 2016-08-23:
I am managing our groups compute server, and I wanted to make the bftools available to all users. So I put the whole bftools folder in the `/opt` folder (I'm on Linux).
Then I symlinked all executables to `/usr/local/bin` such that 
     
    /usr/local/bin/bfconvert -> /opt/bftools/bfconvert

Now all users of this server can simply run bfconvert by typing `bfconvert`, however, this won't work because the `bfconvert` executable looks in the `BF_DIR` for `bf.sh` and 

    BF_DIR=`dirname "$0"`

which is `/usr/local/bin` now, and `bf.sh` is in `/opt/bftools`.

It might be nice to let the scripts resolve to the correct (symlink-resolved) directory, some suggestions are given [here](http://stackoverflow.com/questions/7665/how-to-resolve-symbolic-links-in-a-shell-script).

I could make a pull request for a Linux solution, but I don't have OS X to test it....

sbesson, 2016-08-23:
@burgera: thanks for your suggestion which certainly sounds sensible. Please feel free to open a Pull Request against the repository with the proposed Linux implementation. We certainly do have the infrastructure to test it equally under Linux or OSX.

burgerga, 2016-08-23:
Would something like this be acceptable:
```
OS=`uname -s`
if [[ $OS == "Linux" ]]; then
  BF_DIR=$(dirname $(readlink -f "$0"))
elif [[ $OS == "Darwin" ]]; then
  BF_DIR=$(dirname $(perl -MCwd -le 'print Cwd::abs_path(shift)' "$0"))
else
  BF_DIR=$(dirname "$0")
fi
```

Unfortunately OS X doesn't implement `readlink -f` and `readlink` without `-f` only resolves a single link. Since perl is installed by default on OS X this might be the best solution. On Cygwin (and others I don't know about) you get the `else` case, which is the old behavior.

sbesson, 2016-08-24:
@burgerga: thanks for the investigation and the proposal. Looking briefly at various workaround, we also found this [platform-independent solution](http://serverfault.com/questions/40144/how-can-i-retrieve-the-absolute-filename-in-a-shell-script-on-mac-os-x/406371#406371) using `pwd -P` which we have tested successfully on OSX, CentOS6 and FreeBSD and might be easier to maintain. Thoughts?

burgerga, 2016-08-24:
@sbesson: can you tell me how you symlinked the executables because for me it's not working. I'm guessing you symlinked the directory and not the executable? Or am I missing something...

`$0` is `/usr/local/bin/bfconvert`, so `"$(dirname "$0")"` should give `/usr/local/bin`, and since `/usr/local/bin` is not symlinked, the `pwd -P` will do nothing I guess?

sbesson, 2016-08-24:
@burgerga: of course you are right, I tested the `pwd -P` solution by symlinking the directory and not the file itself. Sorry about the confusion.

So your Perl implementation in https://github.com/openmicroscopy/bioformats/issues/2540#issuecomment-241813764 might indeed be our best solution for now. My main concern if whether we should make the decision based on the `uname -s` output or whether we should try to detect if the executable exists. We can discuss this on the Pull Request if you want to open one.

burgerga, 2016-08-25:
No problem, I asked because I was also hoping for a simpler solution.
I'm working on the pull request, but something else: how is the `bftools.zip` generated? 
Because in the source I see `rwxr-xr-x` permissions (eg `bfconvert`), but in the `bftools.zip` it's `rwxr-x--x` (no read permissions for 'other').
Once you move the files to `/opt` they are owned by `root:root` so we kind of need that read permission...

sbesson, 2016-08-25:
The build target for creating the command-line tools executable is at https://github.com/openmicroscopy/bioformats/blob/develop/ant/toplevel.xml#L995 so we should be able to control the file permissions via `filemode`.

mtbc, 2016-08-25:
It just occurred to me to wonder if there's some way in bash to simply try approaches in descending order of preference and take the first one that seems to succeed, e.g., does Darwin `readlink` give a non-zero exit code with `-f`? It might be more portable than a growing conditional (FreeBSD, OpenBSD, NetBSD...) It's just a thought, may well add no value.

burgerga, 2016-08-25:
There is, you can do a conditional on the exit status (`$?`), but since I don't have a mac I can't test readlink. Seems like a good idea.

sbesson, 2016-08-25:
@burgerga: briefly tested on OSX

```
sbesson@ls30630:tmp $ cat test.bash
readlink -f /tmp > /dev/null 2> /dev/null
if [ $? -ne 0 ]; then
  echo 'failure';
else
  echo 'success'
fi
sbesson@ls30630:tmp $ uname -s
Darwin
sbesson@ls30630:tmp $ bash test.bash
failure
```

On CentOS

```
[sbesson@ome-c6100-1 ~]$ uname -s
Linux
[sbesson@ome-c6100-1 ~]$ bash test.bash 
success
```

On FreeBSD

```
$ uname -s
FreeBSD
$ bash test.bash 
success
```

Let us know if you have a specific implementation you need tested or push to your branch which will be included our Continous Integration daily build.

burgerga, 2016-08-26:
I tried it, but it didn't look that pretty, in the end I found an approach that works with the bash `||`
in which the RHS doesn't get executed if the LHS succeeds. So I condensed it down to the following:

```
RESOLVED_PATH=$(readlink -f "$0" 2>/dev/null \
  || perl -MCwd -le 'print Cwd::abs_path(shift)' "$0" 2>/dev/null \
  || echo "$0")
BF_DIR=$(dirname $RESOLVED_PATH)
```

Which is flexible and doesn't look to complicated (I think ;) )
The `2> /dev/null` is necessary to avoid error messages on the command line if it fails.

sbesson, 2016-08-26:
@burgerga: looks good to me and seems to work as expected on OSX:

```
sbesson@lifesci-50798:tmp $ ls -alh test2
total 8
drwxr-xr-x   3 sbesson  wheel   102B 26 Aug 20:27 .
drwxrwxrwt  13 root     wheel   442B 26 Aug 20:27 ..
lrwxr-xr-x   1 sbesson  wheel    13B 26 Aug 20:27 bf.sh -> ../test/bf.sh
sbesson@lifesci-50798:tmp $ cat test/bf.sh 
RESOLVED_PATH=$(readlink -f "$0" 2>/dev/null \
  || perl -MCwd -le 'print Cwd::abs_path(shift)' "$0" 2>/dev/null \
  || echo "$0")
BF_DIR=$(dirname $RESOLVED_PATH)
echo $BF_DIR
sbesson@lifesci-50798:tmp $ bash test/bf.sh 
/private/tmp/test
sbesson@lifesci-50798:tmp $ bash test2/bf.sh 
/private/tmp/test
```

Do you want to update your PR with this condensed form and we give it a try next week on a set of platforms?

burgerga, 2016-08-26:
Yes, I'll update the PR sometime this weekend

burgerga, 2016-08-27:
Updated and rebased

PR #2547 (2016-08-30): Make bftools symlink-able
Resolves #2540
closed 2016-08-30 by sbesson