ome/bioformats #2544: NDPI to TIFF conversion is failling

felixveysseyre, 2016-08-24:
Hello,

I am trying to convert an NDPI image into TIFF using _bfconvert_ (5.1.10) on Linux (same thing on Mac OS X).

Unfortunately, I get the following error:
```
felixveysseyre$ ~/Softwares/bftools-old/bfconvert -bigtiff -series 0  ~/Downloads/wwj-2.ndpi ~/Downloads/wwj-2.tiff
*** A new stable version is available. ***
*** Install the new version using:     ***
***   'upgradechecker -install'        ***
/Users/felixveysseyre/Downloads/wwj-2.ndpi
NDPIReader initializing /Users/felixveysseyre/Downloads/wwj-2.ndpi
Reading IFDs
Populating metadata
Populating OME metadata
[Hamamatsu NDPI] -> /Users/felixveysseyre/Downloads/wwj-2.tiff [Tagged Image File Format]
Tile size = 1024 x 1024
Exception in thread "main" java.io.IOException
	at loci.formats.services.JPEGTurboServiceImpl.getTile(JPEGTurboServiceImpl.java:332)
	at loci.formats.services.JPEGTurboServiceImpl.getTile(JPEGTurboServiceImpl.java:238)
	at loci.formats.in.NDPIReader.openBytes(NDPIReader.java:193)
	at loci.formats.FormatReader.openBytes(FormatReader.java:920)
	at loci.formats.ImageReader.openBytes(ImageReader.java:437)
	at loci.formats.tools.ImageConverter.convertTilePlane(ImageConverter.java:680)
	at loci.formats.tools.ImageConverter.convertPlane(ImageConverter.java:617)
	at loci.formats.tools.ImageConverter.testConvert(ImageConverter.java:546)
	at loci.formats.tools.ImageConverter.main(ImageConverter.java:874)
Caused by: java.lang.NegativeArraySizeException
	at org.libjpegturbo.turbojpeg.TJDecompressor.decompress(TJDecompressor.java:338)
	at loci.formats.services.JPEGTurboServiceImpl.getTile(JPEGTurboServiceImpl.java:325)
	... 8 more

```

Here is the image information I got from _showinf_:
```
felixveysseyre$ ~/Softwares/bftools/showinf ~/Downloads/wwj-2.ndpi
*** A new stable version is available. ***
*** Install the new version using:     ***
***   'upgradechecker -install'        ***
Checking file format [Hamamatsu NDPI]
Initializing reader
NDPIReader initializing /Users/felixveysseyre/Downloads/wwj-2.ndpi
Reading IFDs
Populating metadata
Populating OME metadata
Initialization took 0.144s

Reading core metadata
filename = /Users/felixveysseyre/Downloads/wwj-2.ndpi
Series count = 5
Series #0 :
	Image count = 1
	RGB = true (3) 
	Interleaved = true
	Indexed = false (true color)
	Width = 36976
	Height = 35976
	SizeZ = 1
	SizeT = 1
	SizeC = 3 (effectively 1)
	Thumbnail size = 128 x 124
	Endianness = intel (little)
	Dimension order = XYCZT (uncertain)
	Pixel type = uint8
	Valid bits per pixel = 8
	Metadata complete = true
	Thumbnail series = false
	-----
	Plane #0 <=> Z 0, C 0, T 0
```

Weird thing, I was able to convert an other NDPI file without any problem.

Should I assume it is coming from a defect in the image ?

Thanks,

dgault, 2016-08-24:
Hi Félix,

If possible would you be able to upload the file that is causing the error for testing?
You can upload the image at https://www.openmicroscopy.org/qa2/qa/upload/

With Thanks,
David Gault

felixveysseyre, 2016-08-24:
Hi @dgault,

Here is the data: https://www.openmicroscopy.org/qa2/qa/feedback/17312/?token=46160a2e02653365a15d68078849b939

Thanks for your help !

dgault, 2016-08-25:
Hi Félix,

Thank you for submitting the file. I can certainly reproduce the same issue that you are seeing.
From debugging the tile dimensions within the JPEG decoder appear to be incorrect, though Im not yet sure if this is due to an error on our part or if the issue is with the image itself.

I will do some further investigation and keep you updated when I know more.

With Thanks,
David Gault

felixveysseyre, 2016-08-25:
Thanks a lot for your time @dgault !

melissalinkert, 2020-01-31:
Confirmed that this issue is still present with #3505.

usuyama, 2022-06-16:
@melissalinkert @felixveysseyre @dgault  Has anyone had success converting NDPI to TIFF?