ome/bioformats #2551: DataTools not Thread-Safe anymore

dietzc, 2016-08-30:
Hi all,

Some of the methods in https://github.com/openmicroscopy/bioformats/blob/develop/components/formats-common/src/loci/common/DataTools.java are not thread-safe anymore since openmicroscopy/bioformats@87730bd.

Reproducible with e.g. calling https://github.com/openmicroscopy/bioformats/blob/develop/components/formats-gpl/src/loci/formats/in/ZeissCZIReader.java#L2499 from two threads at the same time.

Thanks!

Christian

dietzc, 2016-08-30:
To be more precise. https://github.com/openmicroscopy/bioformats/blob/87730bdcf8677d0e889f0e20c2c0d4278ddebe6e/components/formats-common/src/loci/common/DataTools.java#L61 is the problem.

dgault, 2016-08-31:
A PR to address this has been opened - https://github.com/openmicroscopy/bioformats/pull/2552

closed 2016-09-05 by sbesson

sbesson, 2016-09-14:
@dietzc I am sure you saw it, Bio-Formats 5.2.2 is now released and should include the regression fix. Let us know if you still encounter issues on this front and many thanks for your quick report.

dietzc, 2016-09-14:
@sbesson thank you very much :+1: Keep up the great work!