ome/bioformats #2555: Problem with reading Metadata when using BioFormats 5.2.1

sebi06, 2016-09-02:
Hi guys,

i have a test CZI which I use regularly to test the latest BioFormats library from python using python-bioformats. When I run my test script using the bioformats_package.jar version 5.1.10 everything works fine --> see attached out inside BF_5.2.10_OK.txt.

When I try the same using the newer jar version 5.2.1 I get many warnings & co regarding Slidebook and finally an error related to some metainformation --> see attached out inside BF_5.2.10_Error.txt

To me it looks like the way some metainformation are read has changed etc., so that the new OME-XML produced by the new 5.2.1 version now looks different and test scripts does not find the required information any more. Any ideas, what the problem might be?

(by the way, 5.2.0 also shows this problem)

Cheers,

Sebi (from Zeiss) 

Testfile: 

https://dl.dropboxusercontent.com/u/623476/B4_B5_S%3D8_4Pos_perWell_T%3D2_Z%3D1_CH%3D1.czi

[BF_5.1.10_OK.txt](https://github.com/openmicroscopy/bioformats/files/451636/BF_5.1.10_OK.txt)
[BF_5.2.1_Error.txt](https://github.com/openmicroscopy/bioformats/files/451638/BF_5.2.1_Error.txt)

sbesson, 2016-09-02:
Hi @sebi06,

I assume the code you are running is the sane as https://github.com/sebi06/BioFormatsRead with only `bioformats_package` updated to the 5.2.1 version. In addition to API changes, Bio-Formats 5.2.0 also uses the most recent version of the OME Data model (see the [version history](https://www.openmicroscopy.org/site/support/bio-formats5.2/about/whats-new.html#august-18) for full details).
The Slidebook information is logging at the debug level and irrelevant to the underlying error here. Looking at https://github.com/sebi06/BioFormatsRead/blob/master/bfimage/bftools.py#L720, my suspicion is that you need to bump the namespace URL to use `2016-06` instead of `2015-01`. Can you try and let us know if this fixes the issue?

Best,
Sebastien

sebi06, 2016-09-02:
> BioFormatsRead/blob/master/bfimage/bftools.py#L720, my suspicion is that

Hi Sebastian,

Perfect. Changing the namespace URL solved the issue. Sorry i missed this.

Sebi

sbesson, 2016-09-02:
No worries, glad to hear the transition from 5.1.10 to 5.2.0 went smoothly otherwise.

closed 2016-09-02 by sbesson