ome/bioformats #2558: Problem with Reading Image Series Dimension with BioFormats 5.2.1

sebi06, 2016-09-04:
Hi Guys,

i test one of my testfiles yesterday with the BioFormats 5.2.1 and ran into an issue I do not understand yet. The file is a 3x3 tile image with 10% overlap and just one channel, where a frame is 640x640. Wirh former bioformtas this resulted always in 9 image series (one for each tile) with 640x640.
Now I wanted to check out the new features, where my expectation is that it results in only one image series with 1792 x 1792 size.

But with BioFormats 5.2.1 and my testscripts now give me 2 image series, which does not really makes sense to me. Is this a bug or am I missing somethig?
With 5.1.10 everything works normal and I get only one image series

Testfile: https://dl.dropboxusercontent.com/u/623476/3x3Tile_CH%3D1.czi

Help i really appreciated,
Sebi

Here is the output for the testfile using BF 5.1.10 vs. 5.2.1.

[3x3Tile_CH=1_BF5.1.10_output.txt](https://github.com/openmicroscopy/bioformats/files/453818/3x3Tile_CH.1_BF5.1.10_output.txt)
[3x3Tile_CH=1_BF5.2.1_output.txt](https://github.com/openmicroscopy/bioformats/files/453819/3x3Tile_CH.1_BF5.2.1_output.txt)

sbesson, 2016-09-05:
Hi @sebi06, what you observe is expected as perhttps://github.com/openmicroscopy/bioformats/pull/2481 whichadded support for the whole slide imaging modality in the Zeiss CZI format.
As described in the Pull Request, the implications of these support addition for existing filesets are that:
- all 640x640 tiles are now combined into a single full-resolution image of dimensions 1792 x 1792 pixels instead of 9 series/images previously
- a second series for the pyramid at lowest resolution with dimensions 896 x 896 pixels is also detected

In order to navigate through the pyramid sub-resolutions, you want to adapt your test script to use the Bio-Formats resolution API - see https://github.com/openmicroscopy/bioformats/blob/v5.2.1/components/formats-gpl/utils/SubResolutionExample.java for an example.

Best,
Sebastien

sebi06, 2016-09-05:
> formats-gpl/utils/SubResolutionExample.java for an example.

Hi Sebastien,

ok, I got it. So then I think the problem is that python_bioformats (at
least to my knowledge) offers no parameter getResolutionCount. Probably
there are not aware of the sub-resolution API.

I python the following does not work yet:

rdr = bioformats.get_image_reader(None, path=imagefile)

totalseries = np.int(rdr.rdr.getSeriesCount())

for sc in range(0, totalseries):
    rdr.rdr.setSeries(sc)
    resolutionCount = rdr.rdr.getResolutionCount()

So would you suggest to contact those people regarding the problem. The
python wrapper for BioFormats is an extremly useful and powerful tool, we
must sort this out.

Cheers,

Sebi

sebi06, 2016-09-05:
Additionally I am wondering how the user can figure out programmatically which image series contains the full resolution. In ZEN it is easily possible to do things like creating an acquisition with 2 distinct positions, where the 1st position is a 3x3 tile region and the 2nd position is a 9x9 tile region (as an example). This resulted in 8 image series for the test CZI file, which can be found here:

https://dl.dropboxusercontent.com/u/623476/Pos%3D2_3x3_Tile%2B9x9Tile_CH%3D1.czi

In ZEN we call this "2 scenes" and we have a special index for this called SceneIndex. For the test CZI above this would yield in:

Dimension Order CZI  :  BSCYX0
Shape CZI                    :  1, 2, 1, 10243, 9567, 1

How is BioFormats 5.2.1 dealing with this issue?

Cheers, Sebi

sbesson, 2016-09-06:
@sebi06: so using your example file, the behavior depends on whether subresolutions are flattened as series or not. In the former case, 8 series will be detected (one for each scene and resolution):

```
$ showinf -nopix -nometa Pos\=2_3x3_Tile+9x9Tile_CH\=1.czi 
Checking file format [Zeiss CZI]
Initializing reader
ZeissCZIReader initializing Pos=2_3x3_Tile+9x9Tile_CH=1.czi
Initialization took 0.377s

Reading core metadata
filename = Pos=2_3x3_Tile+9x9Tile_CH=1.czi
Series count = 8
Series #0 :
	Image count = 1
	RGB = false (1) 
	Interleaved = false
	Indexed = false (false color)
	Width = 1792
	Height = 1792
	SizeZ = 1
	SizeT = 1
	SizeC = 1
	Thumbnail size = 128 x 128
	Endianness = intel (little)
	Dimension order = XYCZT (certain)
	Pixel type = uint8
	Valid bits per pixel = 8
	Metadata complete = false
	Thumbnail series = false
Series #1 :
	Image count = 1
	RGB = false (1) 
	Interleaved = false
	Indexed = false (false color)
	Width = 896
	Height = 896
	SizeZ = 1
	SizeT = 1
	SizeC = 1
	Thumbnail size = 128 x 128
	Endianness = intel (little)
	Dimension order = XYCZT (certain)
	Pixel type = uint8
	Valid bits per pixel = 8
	Metadata complete = false
	Thumbnail series = false
Series #2 :
	Image count = 1
	RGB = false (1) 
	Interleaved = false
	Indexed = false (false color)
	Width = 448
	Height = 448
	SizeZ = 1
	SizeT = 1
	SizeC = 1
	Thumbnail size = 128 x 128
	Endianness = intel (little)
	Dimension order = XYCZT (certain)
	Pixel type = uint8
	Valid bits per pixel = 8
	Metadata complete = false
	Thumbnail series = false
Series #3 :
	Image count = 1
	RGB = false (1) 
	Interleaved = false
	Indexed = false (false color)
	Width = 224
	Height = 224
	SizeZ = 1
	SizeT = 1
	SizeC = 1
	Thumbnail size = 128 x 128
	Endianness = intel (little)
	Dimension order = XYCZT (certain)
	Pixel type = uint8
	Valid bits per pixel = 8
	Metadata complete = false
	Thumbnail series = false
Series #4 :
	Image count = 1
	RGB = false (1) 
	Interleaved = false
	Indexed = false (false color)
	Width = 5248
	Height = 5248
	SizeZ = 1
	SizeT = 1
	SizeC = 1
	Thumbnail size = 128 x 128
	Endianness = intel (little)
	Dimension order = XYCZT (certain)
	Pixel type = uint8
	Valid bits per pixel = 8
	Metadata complete = false
	Thumbnail series = false
Series #5 :
	Image count = 1
	RGB = false (1) 
	Interleaved = false
	Indexed = false (false color)
	Width = 2624
	Height = 2624
	SizeZ = 1
	SizeT = 1
	SizeC = 1
	Thumbnail size = 128 x 128
	Endianness = intel (little)
	Dimension order = XYCZT (certain)
	Pixel type = uint8
	Valid bits per pixel = 8
	Metadata complete = false
	Thumbnail series = false
Series #6 :
	Image count = 1
	RGB = false (1) 
	Interleaved = false
	Indexed = false (false color)
	Width = 1312
	Height = 1312
	SizeZ = 1
	SizeT = 1
	SizeC = 1
	Thumbnail size = 128 x 128
	Endianness = intel (little)
	Dimension order = XYCZT (certain)
	Pixel type = uint8
	Valid bits per pixel = 8
	Metadata complete = false
	Thumbnail series = false
Series #7 :
	Image count = 1
	RGB = false (1) 
	Interleaved = false
	Indexed = false (false color)
	Width = 656
	Height = 656
	SizeZ = 1
	SizeT = 1
	SizeC = 1
	Thumbnail size = 128 x 128
	Endianness = intel (little)
	Dimension order = XYCZT (certain)
	Pixel type = uint8
	Valid bits per pixel = 8
	Metadata complete = false
	Thumbnail series = false
```

If resolutions are not flattened then 2 series will be detected (one per scene) containing 4 levels of resolution each.

```
$ showinf -nopix -nometa -noflat Pos\=2_3x3_Tile+9x9Tile_CH\=1.czi 
Checking file format [Zeiss CZI]
Initializing reader
ZeissCZIReader initializing Pos=2_3x3_Tile+9x9Tile_CH=1.czi
Initialization took 0.35s

Reading core metadata
filename = Pos=2_3x3_Tile+9x9Tile_CH=1.czi
Series count = 2
Series #0 :
	Resolutions = 4
		sizeX[0] = 1792
		sizeX[1] = 896
		sizeX[2] = 448
		sizeX[3] = 224
	Image count = 1
	RGB = false (1) 
	Interleaved = false
	Indexed = false (false color)
	Width = 1792
	Height = 1792
	SizeZ = 1
	SizeT = 1
	SizeC = 1
	Thumbnail size = 128 x 128
	Endianness = intel (little)
	Dimension order = XYCZT (certain)
	Pixel type = uint8
	Valid bits per pixel = 8
	Metadata complete = false
	Thumbnail series = false
Series #1 :
	Resolutions = 4
		sizeX[0] = 5248
		sizeX[1] = 2624
		sizeX[2] = 1312
		sizeX[3] = 656
	Image count = 1
	RGB = false (1) 
	Interleaved = false
	Indexed = false (false color)
	Width = 5248
	Height = 5248
	SizeZ = 1
	SizeT = 1
	SizeC = 1
	Thumbnail size = 128 x 128
	Endianness = intel (little)
	Dimension order = XYCZT (certain)
	Pixel type = uint8
	Valid bits per pixel = 8
	Metadata complete = false
	Thumbnail series = false
```

On the python-bioformats front, looking at the code, it looks like every method needs to be exposed via the Java bridge. Opening an issue on their repository to discuss the API addition might be a starting point as I do not know about theor roadmap.

sbesson, 2016-09-13:
See https://github.com/CellProfiler/python-bioformats/issues/61 for the corresponding `python-bioformats` issue.

closed 2016-09-13 by sbesson