ome/bioformats #2575: Problem with getImageID in MATLAB using BioFormats 5.1.10

sebi06, 2016-09-13:
Hi guys,

I tried the following in MATLAB

% To access the file reader without loading all the data, use the low-level bfGetReader.m function:
reader = bfGetReader(filename);

% You can then access the OME metadata using the getMetadataStore() method:
omeMeta = reader.getMetadataStore();

% get ImageCount --> currently only reading one image is supported
imagecount = omeMeta.getImageCount();
imageID = omeMeta.getImageID()

But I get the error:

No method 'getImageID' with matching signature found for class
'loci.formats.ome.OMEXMLMetadataImpl'.

But the method is offered by the auto-completion but it does nott work. What I am missing?

Sebi

sbesson, 2016-09-13:
Hi @sebi06, for most of the Image getter methods of the `MetadataRetrieve` interface, you need to pass the index of the image, e.g. for [getImageID(int)](http://downloads.openmicroscopy.org/bio-formats/5.1.10/api/ome/xml/meta/MetadataRetrieve.html#getImageID(int))

```
imageID = omeMetadata.getImageID(0)
```

sebi06, 2016-09-14:
Hi,

now i look like a idiot ... :-) I could swear I tried this ....

closed 2016-09-14 by sebi06