ome/bioformats #2576: Include version number in btools zip

burgerga, 2016-09-15:
Hi I saw that 5.2.2 was out (even saw my name in release notes ;) )
But at the moment it's a bit tricky to find out which version you're running.

When I run any tool from bioformats I get

```
*** A new stable version is available. ***
*** Install the new version using:     ***
***   'upgradechecker -install'        ***
```

and I don't think the `upgradechecker` is included in the zip.

Also the zip itself and the download website don't tell you which version you will download...

sbesson, 2016-09-15:
Hi @burgerga. As an immediate feature, most of the command-line tools have a `-version` option which will print the version, the sha1 and the build date and exit

```
$ showinf -version
Version: 5.2.2
VCS revision: 00d095359df2ae798a1ccc6c394347a2f0aebd94
Build date: 13 September 2016
```

Agreed on the need for information, the most appropriate file to consider including into the distributed command-line tools might be https://github.com/openmicroscopy/bioformats/blob/develop/docs/sphinx/about/whats-new.txt. Thoughts?

burgerga, 2016-09-15:
Can't believe I overlooked the obvious cli argument, must have been the lack of coffee :P
Yes, the whats-new.txt would be nice, but maybe renamed to the conventional NEWS or CHANGELOG.

sbesson, 2016-09-15:
@burgerga: renaming the file for distribution looks sensible. Feel free to open a Pull Request implementing your proposal directly (NB: the Ant target creating the command-line tools bundle is at https://github.com/openmicroscopy/bioformats/blob/v5.2.2/ant/toplevel.xml#L995) or let us know if we should capture this on our side as a RFE.

burgerga, 2016-09-15:
I can do it, but maybe it's best that you decide what conventions you want to follow. I like the conventions from http://keepachangelog.com/ (CHANGELOG) but maybe you like GNU (NEWS) conventions better?

simleo, 2016-09-16:
@burgerga @sbesson note that this is not simply a rename. Rather, we need to add a new plain text redering for `whats-new.txt`. See https://trello.com/c/mRvqiTsP/46-add-news-file

rleigh-codelibre, 2016-09-16:
https://github.com/openmicroscopy/bioformats/pull/2581 is a PR to tackle the first point: the version number of the zip file.

burgerga, 2016-09-19:
I can't post on trello, but I don't see a problem including the rst file as is, most of the content should be clear. (PyPi also uses rst for READMEs (offtopic: although people are slightly annoyed that markdown isn't supported https://github.com/pypa/packaging-problems/issues/46)

sbesson, 2016-09-19:
@burgerga: as you may read on the Trello card, copying the rst file as its is is exactly the approach used by the Octave package (https://github.com/openmicroscopy/bioformats/blob/v5.2.2/ant/toplevel.xml#L1045). So :+1: for applying this strategy to the command-line tools bundle in a first step from our side.

sbesson, 2016-09-22:
Closing as per https://github.com/openmicroscopy/bioformats/pull/2584 which will be included in the upcoming Bio-Formats 5.2.3 release.

closed 2016-09-22 by sbesson