ome/bioformats #2577: Problem with nd2 -> tif conversion

burgerga, 2016-09-15:
I have an nd2 file here that I'm converting to tifs, however the tiffs that come out seem to be missing something. The data is all there, and Fiji (using BioFormats) is opening it fine, however, Nautilus (I'm on linux) is not generating a preview, and Ilastik can't seem to open them (spits out assertion errors).

If I open the file in Fiji and then save it as tiff, all problems seem to go away; I do get a preview in Nautilus and it opens in Ilastik just fine.

I use the following macro in Fiji to convert the image
```
run("Bio-Formats Windowless Importer", "open=" + file);
saveAs("Tiff", file);
close();
```
I'v attached the two tiff files, and if needed it's no problem to send you the nd2 file (11GB).

Tested on bioformats 5.2.2 and 5.1.10

[bioformats_issue2577.zip](https://github.com/openmicroscopy/bioformats/files/474301/bioformats_issue2577.zip)

rleigh-codelibre, 2016-09-15:
See the difference in the TIFFs below.  The first has an colour map (palette), which is copied from the original ND2 file, while the second has not and is a plain greyscale image.  In both cases the pixel data is identical; the difference lies solely in the addition of the colour map.

The problem you are seeing is that Nautilus isn't handling TIFFs with a colour map properly.  It's likely just giving up if the photometric interpretation is PALETTE.

Note that with Bio-Formats 5.2.2, `bfconvert` gained a `-nolookup` option which will disable the conversion of lookup tables, should you wish to omit them.

Regards,
Roger

```
% tiffinfo tif_from_bf.tif
TIFF Directory at offset 0x8 (8)
  Image Width: 512 Image Length: 512
  Resolution: 16279.4, 16279.4 pixels/cm
  Bits/Sample: 16
  Sample Format: unsigned integer
  Compression Scheme: None
  Photometric Interpretation: palette color (RGB from colormap)
  Samples/Pixel: 1
  Rows/Strip: 1
  Planar Configuration: single image plane
  Color Map: (present)
  ImageDescription: ImageJ=
hyperstack=true
images=1
channels=1
slices=1
frames=1
  Software: OME Bio-Formats 5.2.2

% tiffinfo saved_with_fiji.tif
TIFFReadDirectory: Warning, Unknown field with tag 50838 (0xc696) encountered.
TIFFReadDirectory: Warning, Unknown field with tag 50839 (0xc697) encountered.
TIFF Directory at offset 0x8 (8)
  Subfile Type: (0 = 0x0)
  Image Width: 512 Image Length: 512
  Resolution: 16279.4, 16279.4 (unitless)
  Bits/Sample: 16
  Compression Scheme: None
  Photometric Interpretation: min-is-black
  Samples/Pixel: 1
  Rows/Strip: 512
  Planar Configuration: single image plane
  ImageDescription: ImageJ=1.51f
unit=pixel
min=31.0
max=563.0

  Tag 50838: 12,64
  Tag 50839: 73,74,73,74,105,110,102,111,0,0,0,1,0,83,0,111,0,102,0,116,0,119,0,97,0,114,0,101,0,58,0,32,0,79,0,77,0,69,0,32,0,66,0,105,0,111,0,45,0,70,0,111,0,114,0,109,0,97,0,116,0,115,0,32,0,53,0,46,0,50,0,46,0,50,0,10
```

burgerga, 2016-09-15:
Thanks for your reply, using `-nolookup` indeed solves all the issues I'm having. I'll update to 5.2.2. Is this palette such an unusual feature that various programs don't know how to handle it?

rleigh-codelibre, 2016-09-15:
It's not *that* unusual, but it may well be the case that programs know how to deal with 8-bit lookup tables, but not 16-bit.  We added the `-nolookup` option a few weeks back in response to another report of this being problematic.  We have been creating the lookup tables for some time (16-bit since 2013 with bfconvert), but for 5.2 some readers now expose their lookup tables which previously did not.  So while we are technically more correct, it is a behaviour change which in some cases causes problems with tools which have defects in their TIFF support, and we added `-nolookup` to cater for that.

For 5.3.0 we will likely review the defaults here, and maybe make lookup table conversion opt-in rather than opt-out if the cost/benefit makes it worth switching.

Regards,
Roger

burgerga, 2016-09-15:
Thanks again for your informative answer!

closed 2016-09-15 by burgerga

rleigh-codelibre, 2016-09-15:
By the way, if you did want to investigate what Nautilus is doing, it's likely as simple as a missing `case` for that photometric interpretation/pixel size combination.  If it can already handle 8-bit lookup tables and 8-bit and 16-bit images, it can probably be tweaked to support 16-bit lookup tables with relative ease.  It's certainly all supported by the libtiff library; it's probably at the application level it's missing the special case.