ome/bioformats #2578: Fiji fails to start

PierreMuller, 2016-09-15:
After update of today, that only contained bioformat jar changes,
Fiji is crashing at startup.

I tried to use replace bioformats-plugin.jar by bioformats_package.jar
but it still fails.

Pierre Muller

[DEBUG] Created service 'io.scif.io.DefaultNIOService' in 0 ms
[DEBUG] Creating service: io.scif.ome.services.DefaultOMEMetadataService
Error while executing the main() method of class 'net.imagej.Main':
java.lang.IllegalArgumentException: Invalid service: io.scif.ome.services.DefaultOMEMetadataService
        at org.scijava.service.ServiceHelper.createExactService(ServiceHelper.java:280)
        at org.scijava.service.ServiceHelper.loadService(ServiceHelper.java:231)
        at org.scijava.service.ServiceHelper.loadService(ServiceHelper.java:194)
        at org.scijava.service.ServiceHelper.loadServices(ServiceHelper.java:166)
        at org.scijava.Context.<init>(Context.java:277)
        at org.scijava.Context.<init>(Context.java:234)
        at org.scijava.Context.<init>(Context.java:174)
        at org.scijava.Context.<init>(Context.java:160)
        at net.imagej.ImageJ.<init>(ImageJ.java:79)
        at net.imagej.Main.launch(Main.java:61)
        at net.imagej.Main.main(Main.java:73)
        at sun.reflect.NativeMethodAccessorImpl.invoke0(Native Method)
        at sun.reflect.NativeMethodAccessorImpl.invoke(NativeMethodAccessorImpl.java:39)
        at sun.reflect.DelegatingMethodAccessorImpl.invoke(DelegatingMethodAccessorImpl.java:25)
        at java.lang.reflect.Method.invoke(Method.java:597)
        at net.imagej.launcher.ClassLauncher.launch(ClassLauncher.java:279)
        at net.imagej.launcher.ClassLauncher.run(ClassLauncher.java:186)
        at net.imagej.launcher.ClassLauncher.main(ClassLauncher.java:77)
Caused by: java.lang.UnsupportedClassVersionError: ome/xml/model/enums/EnumerationException : Unsupported major.minor version 51.0
        at java.lang.ClassLoader.defineClass1(Native Method)
        at java.lang.ClassLoader.defineClassCond(ClassLoader.java:632)
        at java.lang.ClassLoader.defineClass(ClassLoader.java:616)
        at java.security.SecureClassLoader.defineClass(SecureClassLoader.java:141)
        at java.net.URLClassLoader.defineClass(URLClassLoader.java:283)
        at java.net.URLClassLoader.access$000(URLClassLoader.java:58)
        at java.net.URLClassLoader$1.run(URLClassLoader.java:197)
        at java.security.AccessController.doPrivileged(Native Method)
        at java.net.URLClassLoader.findClass(URLClassLoader.java:190)
        at java.lang.ClassLoader.loadClass(ClassLoader.java:307)
        at sun.misc.Launcher$AppClassLoader.loadClass(Launcher.java:301)
        at java.lang.ClassLoader.loadClass(ClassLoader.java:248)
        at java.lang.Class.getDeclaredConstructors0(Native Method)
        at java.lang.Class.privateGetDeclaredConstructors(Class.java:2389)
        at java.lang.Class.getConstructor0(Class.java:2699)
        at java.lang.Class.newInstance0(Class.java:326)
        at java.lang.Class.newInstance(Class.java:308)
        at org.scijava.service.ServiceHelper.createServiceRecursively(ServiceHelper.java:302)
        at org.scijava.service.ServiceHelper.createExactService(ServiceHelper.java:269)
        ... 17 more

sbesson, 2016-09-15:
@PierreMuller: from the stack trace it looks like this might be related to SCIFIO rather than Bio-Formats and `Unsupported major.minor version 51.0` suggests it is a JDK compatibilty issue. Could you paste here and in the Fiji thread which Java is used and which update sites were enabled?

PierreMuller, 2016-09-15:
Hi, 
in top part of debug.exe output, I see:
Detected ImageJ2
Using JRE from ImageJ.cfg: C:\Users\Pierre\FIJI-W~1\Fiji.app/jre
Found Java options in ImageJ.cfg: '-Xmx6127m -cp ij.jar'
JRE not found in 'C:\Users\Pierre\FIJI-W~1\Fiji.app/jre'
JAVA_HOME contains a JRE: 'C:\Users\Pierre\FIJI-W~1\Fiji.app/java/win64/jdk1.6.0_24//jre'

I don't know how to get the exact version from javaw.exe ...

Regarding your second question, I do remember that I added a "specific site" for updates
inside FiJi, but here again, I do not know where to look for it.

Could you tell me where I can find these informations?

Thanks in advance,

Pierre Muller

PierreMuller, 2016-09-15:
Found version command-line option:
-version.
It only works with java.exe (ont with javaw.exe),
but I assume that they are the sazmeversion in the same directory.

The version I was using is 
java version "1.6.0_24"
Java(TM) SE Runtime Environment (build 1.6.0_24-b07)
Java HotSpot(TM) 64-Bit Server VM (build 19.1-b02, mixed mode)

I had a more recent one:
"C:\Program Files\Java\jre1.8.0_101\bin\java.exe" -version
java version "1.8.0_101"
Java(TM) SE Runtime Environment (build 1.8.0_101-b13)
Java HotSpot(TM) 64-Bit Server VM (build 25.101-b13, mixed mode)

Using the javaw.exe from this directory did indeed solve the start failure.

In the java debug console, there are still a lot of DEBUG messages
concerning the org.scijava.Context, but I have no idea if this
is normal or not.

  It seems that the FiJI shipped jre 1.6 is indeed incompatible with recent bioformat changes.

Anyway, I can start FiJi again :-)
Thanks for the help,

Pierre Muller

sbesson, 2016-09-15:
Glad to hear it and thanks for the heads up.  Can you also let the Fiji people know it has been resolved? Closing this issue.

closed 2016-09-15 by sbesson