ome/bioformats #2613: MetamorphReader reads data from unrelated tif files

imagejan, 2016-10-13:
When there are additional files in a folder containing a Metamorph dataset (`nd` file + several `stk` files), it can happen that the Reader gets confused by tif images having a similar name. I uploaded `nd_issue.zip` via http://qa.openmicroscopy.org.uk/qa/upload/ to reproduce the issue (sorry, I must have missed how to keep track of the ticket number there).

[image]

With the tif file present, opening the nd file in Fiji results in entirely white preview thumbs in the series dialog, and in some artifacts in the opened image, apparently the pixel intensity scaling is incorrect and gets wrapped around the UnsignedShort limit.

Renaming the `.tif` file to `.tmp` restores the correct, expected behavior.

[image]

I assume there's something wrong in how the reader is looking for the STK files here:

https://github.com/openmicroscopy/bioformats/blob/fff8e47a3a3490343005b7b26d0158b5075ef01e/components/formats-gpl/src/loci/formats/in/MetamorphReader.java#L1345-L1406

dgault, 2016-10-18:
Hi Jan,

Thanks for providing the sample fileset for testing this issue. I have been able to reproduce the problem just as you described and have spent time debugging through the code to locate the cause. 

It appears that the reader is able to locate the stk files successfully with and without the tif present but it also sees the tif file as part of the fileset. Was the tif supposed to be part of the same fileset with the nd and stk files? Or was it simply a similarly named file in the same location?

imagejan, 2016-10-18:
Hi David,

In the real case that I used to compile the above example dataset, we have a series of (similarly named) nd datasets that are being stitched using Fiji and the Grid/Collection Stitching plugin (using bioformats in the backend). For some of the datasets, we write multi-channel tiff files with Fiji for each tile (=set of one stk file per channel) to improve stitching quality and avoid having to stitch the channels separately one after each other.

So, in short, the tif file(s) were not supposed to belong to the dataset, but were derived from one of the other nd/stk datasets in the same folder.

Does that make sense?

dgault, 2016-10-18:
Yeah that makes total sense in that scenario. The naming of the tiff files will likely be an issue though. If the MetamorphReader is used to open the nd file, then the reader will judge any file that starts with the same name and is either nd, stk, tif or tiff to be included as part of the same fileset. If it is possible to rename the tiff files with a prefix at the start to identify them then that would also resolve the problem when reading the nd file.

mtbc, 2016-10-19:
QA ticket is 17379.

imagejan, 2016-10-19:
> the reader will judge any file that starts with the same name and is either nd, stk, tif or tiff to be included as part of the same fileset.

Well, in this case the file clearly doesn't start with the _same name_, as the basename part of the dataset is `ScanSlide_exp1_rep1` whereas the tiff file starts with `ScanSlide_exp1_rep0`, which is AFAICT a different name according to the [file naming conventions](http://mdc.custhelp.com/app/answers/detail/a_id/18979) which define only `_s`, `_w` and `_t` followed by digits as special parts of the name.

dgault, 2016-10-20:
Hi Jan, to clarify the 'starts with the same name' comment, it is correct that the reader will identify its list of stk files based on the full name of the nd file followed by _s, _w and _t. However when it is first initialised it scans files in the parent directory and will accept any that start with the same prefix (the name of the nd file upto the first underscore character) in this case simply ScanSlide, that have a suffix of stk, tif or tiff. 

I would have to test further with our repository of sample files to check if there is a particular use case for using this shorter naming test.

imagejan, 2016-10-25:
Thanks for the clarification, @dgault.
I don't think that defining the prefix up to the first underscore (`_`) only is covered by the above cited naming conventions. If you have a dataset in your repository where this paradigm is necessary and not covered by the `_[swt]\d+` (regex) rule, I'd be interested to know the software that was writing these files :smile:
I can just say that the current behavior can be very confusing for users in our facility who've been trained to replace spaces by underscores in their filenames (for numerous other reasons).

dgault, 2017-01-10:
Hi Jan,

Just to follow up on this issue of naming conventions. I have been able to spend some further time investigating and checking our repository for datasets which would not match upto the `_[swt]\d+` (regex) rule. We do have some examples of filesets which would fit this category, but they are either .nd and .tif or .nd and .stk, we do not have any examples such as yours which contains a combination of all 3.

From my testing it appears that in cases with an .nd file and only one of .tif or .stk filesets, that the correct files are selected irrespective of the use of `name.substring(0, name.indexOf('_'));`.

This particular issue only appears to become problematic when a mix of all 3 file extensions is present. As such this does look to be a bug in the way the fileset is parsed as opposed to the intended behaviour. I have added a card to our Bio-Formats 5.3 inbox to work on addressing this bug.

David

imagejan, 2017-01-10:
@dgault Thanks for the follow-up.

> we do not have any examples such as yours which contains a combination of all 3.

To be clear, each single dataset consists of only (.nd and .stk) or (.nd and .tif) files. The issue appears when multiple acquisitions in 2D (tif) and 3D (stk) get saved to the same folder. According to the nd file specification, it should be clearly defined which files belong to one dataset even if there are other files in the folder, so I agree this is purely a problem of fileset parsing.

Bio-Formats should not get confused by the presence of unrelated files in the same folder.

imagejan, 2017-06-02:
I couldn't find a card on https://trello.com/b/IBHfAIMP/bio-formats-5.x?menu=filter&filter=Metamorph tracking progress of this issue. @dgault can you point me to one if there is one?
A couple of our users keep running into this issue when they have mixed 2D and 3D aquisitions in the same folder. Let me know if you need more examples.

dgault, 2017-06-05:
Hi Jan, 

Apologies for the delayed response, we were busy with our annual users meeting last week. I have opened a test PR to see exactly how many of our current dataset repo would be affected by changing the current behaviour.

imagejan, 2017-12-20:
@dgault any news on this? As far as I can see, the PR was closed due to an erroneous `System.out.println` statement, but the remaining changes were never merged, were they?

joshmoore, 2017-12-21:
Hi @imagejan. I think the timing of the `System.out.println` comment is just a coincidence. Unfortunately, @dgault @sbesson and almost everyone else is away for the holidays. I'll make sure this (& [the related card](https://trello.com/c/MFE6khf6/81-metamorphreader-unrelated-files)) get looked at again as soon as everyone reappears. All the best.

dgault, 2018-01-03:
Hi @imagejan,

My apologies for the delayed response over the holiday period. The previous PR was closed as it was not properly addressing the problem. The main 

I have been further investigating this issue today and main issue we really need to resolve is to be able to correctly identify from the `.nd` file if the associated files are `.stk` or `.tif` format. Taking a look at http://mdc.custhelp.com/app/answers/detail/a_id/18979/kw/nd provides a more up to date spec than the one we previously had been using. It does confirm that a mixture of both stk and tif is valid. It also states that `.stk` indicates a z series was performed and stored in a MetaMorph stack file.

The more relevant part is as follows:
`1 The file type used for Z series data – Stack (*.stk) or the newer Meta Series Single/Multi-plane TIFF (*.tif) – is determined by the option Save Images Using Legacy Format (.stk, old-format .tif). To use this option, from the Edit menu, select Preferences > Image Save Preference. The .ND file records which file type is used:  NDInfoFile Version 1.0 – Legacy format or NDInfoFile Version 2.0 – newer Single/Multi-planeTIF format.`

This would suggest that for Version 2.0 we have only tif files. While for Version 1.0 we can check each wavelength to see if the expected file is .stk or not using `WaveDoZ2", TRUE`.

If this is the case then based on that I should be able to open a new PR shortly which correctly predicts the associated files without as much guess work or relying on pattern files.

imagejan, 2018-01-03:
@dgault wrote:
> This would suggest that for Version 2.0 we have only tif files. While for Version 1.0 we can check each wavelength to see if the expected file is .stk or not using WaveDoZ2", TRUE.

Based on my experience with .nd datasets, I can confirm your assessment: Version 1.0 uses mixed tif/stk files depending on the presence of z stacks for each channel; Version 2.0 uses tif files for both single planes and stacks.

In both versions, the information contained in the .nd file is unambiguous and should not require any guessing/testing with regards to the file names and locations (except to check for missing files when handling incomplete datasets, of course).

Thank you very much for dealing with this, and let me know if you need more sample datasets (particularly Version 1.0 used by VisiView).

dgault, 2018-01-08:
As a follow up to this issue I have opened a new PR which uses the versioning info to remove the ambiguous matching of either stk or tif files - https://github.com/openmicroscopy/bioformats/pull/3032

sbesson, 2018-03-14:
Closing this issue in favor of the opened #3032 (which still requires work to comply with all Metamorph filesets) and the [card](https://trello.com/c/MFE6khf6/81-metamorphreader-unrelated-files) capturing the outstanding task. Thank you  for your contribution, they are much appreciated.

closed 2018-03-14 by sbesson