ome/bioformats #2628: Avoid opening the image attachment for CZI files

sebi06, 2016-10-23:
Hi,

when I use the latest version of bioformats to open a CZI file from within Fiji, it always opens the image along with the image attachment. How can I avoid this?

Most CZI images do not use this attachment, so this should be an option, but not the default. Or is there already a way how to change this behavior?

Cheers, Sebi (from Zeiss)

melissalinkert, 2016-10-24:
Hi Sebi,

Each attachment image is stored as a separate Bio-Formats series, so there is already a way to prevent these images from being displayed.  When opening files in ImageJ/Fiji, the ```Open all series``` box should be unchecked in the ```Bio-Formats Import Options``` window.  In the resulting series chooser (the one with thumbnails), the attachment images should be shown at the very bottom, so just make sure they're unchecked.  If the files you're working with have a large number of series such that the thumbnail window is not shown, then choosing any combination of series except for the last one or two should work in general (depending on how many attachments there are).

If that doesn't help, then we'll need to know a little more about which options you have selected when opening the files.

sebi06, 2016-10-25:
Hi Melissa,

this sounds fine, but I still think there are some open questions here.

Imagine one acquires images from a wellplate, e.g. 96 images (2 Channels) = one per well. This used to be 96 series, which opened as a stack of 96 images usually, when the Open All option was checked. Now this will open the attachment as well and its dimension will be different from all the other image series. Therefore any automatic script that used to work will now have to deal with this one image inside the series, that is "different".

What also puzzle me is the fact that this approach does not really reflect the difference between

- 2 Wells with 9 Positions each --> 18 Series
- 2 Wells with 3x3 Tile Region each --> 18 Series? (by the way, the 2nd approach leads to a error --> separate issue)

Will the label image always be the last inside the whole series?

And currently this label image (BioFormats 5.2.4) returns an empty image with the wrong XY dimensions. This might be a bug.

From a user point of view the additional option to open an label images using a distinct way would be more more convenient in my opinion, because the user either wants to see the real image data or the attachment. To mix those together in one series fells strange to me.

sbesson, 2016-12-13:
Closing as per https://github.com/openmicroscopy/bioformats/issues/2634#issuecomment-266784419

closed 2016-12-13 by sbesson