ome/bioformats #2632: BioFormats 5.2.4 throws series error when opening CZI

sebi06, 2016-10-25:
Hi,

i have a CZI file, where I acquired a 3x3Tile within 50 wells of a 96 plate. When I open this CZI using BioFormats 5.2.4 I get the following error:

(Fiji Is Just) ImageJ 2.0.0-rc-54/1.51g; Java 1.8.0_66 [64-bit]; Windows 7 6.1; 301MB of 12170MB (2%)
 
java.lang.IllegalArgumentException: Invalid series: 100
	at loci.formats.FormatReader.setCoreIndex(FormatReader.java:1368)
	at loci.formats.in.ZeissCZIReader.initFile(ZeissCZIReader.java:974)
	at loci.formats.FormatReader.setId(FormatReader.java:1401)
	at loci.plugins.in.ImportProcess.initializeFile(ImportProcess.java:505)
	at loci.plugins.in.ImportProcess.execute(ImportProcess.java:143)
	at loci.plugins.in.Importer.showDialogs(Importer.java:140)
	at loci.plugins.in.Importer.run(Importer.java:76)
	at loci.plugins.LociImporter.run(LociImporter.java:78)
	at ij.IJ.runUserPlugIn(IJ.java:217)
	at ij.IJ.runPlugIn(IJ.java:181)
	at ij.Executer.runCommand(Executer.java:137)
	at ij.Executer.run(Executer.java:63)
	at java.lang.Thread.run(Thread.java:745)

When I do the same but use 9 Positions instead of a 3x3 Tile for each well, BioFormats 5.2.4 correctly recognizes 450 series.

Cheers, Sebi

Datasets:

https://dl.dropboxusercontent.com/u/623476/96Well_Ch%3D1_50Wells_3x3_pro_Well.czi

https://dl.dropboxusercontent.com/u/623476/96Well_Ch%3D1_50Wells_9P_pro_Well.czi

melissalinkert, 2016-11-07:
This is also now resolved with the current state of https://github.com/openmicroscopy/bioformats/pull/2648.  ``96Well_Ch=1_50Wells_3x3_pro_Well.czi``` now reports 100 series: 2 pyramid resolutions x 50 wells, with all tiles automatically stitched together to match what is shown in Zen.

sebi06, 2016-11-08:
Hi Melissa,

Perfect. And do you know what will happenen when someone applies the
Stitching plugin to such an image?
This will be the typical workflow, but to check this one needs real data
not the testimage from above.

imagejan, 2016-11-08:
I expect that this will require substantial changes to the [Grid/Collection Stitching](https://github.com/fiji/Stitching/blob/master/src/main/java/plugin/Stitching_Grid.java) plugin, as it currently relies on each tile being a separate series in a multiseries file, and it doesn't handle multiple pyramid resolutions yet, to my knowledge.

sebi06, 2016-11-08:
Hi,

as far as I know the Stitching plugin reads the xyz coordinates from the
metadata. Until now one can read a CZI tile image directly and use the
stitching plugin. This worklflow is really important and must still work
with the new BioFormats releases. But of course those tiles CZIs are
currently treated as image series, e.g. a 3x5 tile image is treated as as
15 image series.

And now this will be ... different series representing the pyramidal
resolutions. This will be a serious problem. How should we approach this?
Whom should we contact.

And are the XYZ stage coordinates still to be found in the pyramidal images
at every resolution or at least in the one with the highest resolution?
This would be crucial!

Sebi

sbesson, 2016-12-13:
Closing as per https://github.com/openmicroscopy/bioformats/issues/2634#issuecomment-266784419

closed 2016-12-13 by sbesson