ome/bioformats #2665: Usage of showinf (bftools) does not give the expected output

sebi06, 2016-11-18:
Hi guys,

I converted a CZI file usinsg bfconvert first and then I tried using BioFormats 5.1.10 to test some command line tools. And I have the following problem:

`showinf -no-upgrade -nopix -omexml-only test.ome.tiff`

This give no output inside the CMD tool. I was expecting at least some output, right?

`showinf -no-upgrade -nopix -nocore -omexml-only test.ome.tiff`

This is doing just the same. No output at all.

`showinf -no-upgrade -no-meta -nopix -omexml  -series 0 test.ome.tiff`

This gives me output, but for all series, but the OME-XML part is not shown. Any idea what my mistake is?

Sebi

C:\Users\M1SRH\Documents\Software\BioFormats_Package\5.1.10>showinf -no-upgrade -nometa -nopix -omexml -series 0 l:\Data\BioFormats_CZI_Test\Test_bftools\Test.ome.tiff
Checking file format [OME-TIFF]
Initializing reader
OMETiffReader initializing l:\Data\BioFormats_CZI_Test\Test_bftools\Test.ome.tiff
Reading IFDs
Populating metadata
Initialization took 1.701s

Reading core metadata
filename = l:\Data\BioFormats_CZI_Test\Test_bftools\Test.ome.tiff
Series count = 5
Series #0 -- Test.czi #1:
        Image count = 4
        RGB = false (1)
        Interleaved = false
        Indexed = false (false color)
        Width = 1960
        Height = 1416
        SizeZ = 1
        SizeT = 2
        SizeC = 2
        Thumbnail size = 128 x 92
        Endianness = intel (little)
        Dimension order = XYCZT (certain)
        Pixel type = uint16
        Valid bits per pixel = 14
        Metadata complete = true
        Thumbnail series = false
Series #1 -- Test.czi #2:
        Image count = 4
        RGB = false (1)
        Interleaved = false
        Indexed = false (false color)
        Width = 1960
        Height = 1416
        SizeZ = 1
        SizeT = 2
        SizeC = 2
        Thumbnail size = 128 x 92
        Endianness = intel (little)
        Dimension order = XYCZT (certain)
        Pixel type = uint16
        Valid bits per pixel = 14
        Metadata complete = true
        Thumbnail series = false
Series #2 -- Test.czi #3:
        Image count = 4
        RGB = false (1)
        Interleaved = false
        Indexed = false (false color)
        Width = 1960
        Height = 1416
        SizeZ = 1
        SizeT = 2
        SizeC = 2
        Thumbnail size = 128 x 92
        Endianness = intel (little)
        Dimension order = XYCZT (certain)
        Pixel type = uint16
        Valid bits per pixel = 14
        Metadata complete = true
        Thumbnail series = false
Series #3 -- Test.czi #4:
        Image count = 4
        RGB = false (1)
        Interleaved = false
        Indexed = false (false color)
        Width = 1960
        Height = 1416
        SizeZ = 1
        SizeT = 2
        SizeC = 2
        Thumbnail size = 128 x 92
        Endianness = intel (little)
        Dimension order = XYCZT (certain)
        Pixel type = uint16
        Valid bits per pixel = 14
        Metadata complete = true
        Thumbnail series = false
Series #4 -- Test.czi #5:
        Image count = 4
        RGB = false (1)
        Interleaved = false
        Indexed = false (false color)
        Width = 1960
        Height = 1416
        SizeZ = 1
        SizeT = 2
        SizeC = 2
        Thumbnail size = 128 x 92
        Endianness = intel (little)
        Dimension order = XYCZT (certain)
        Pixel type = uint16
        Valid bits per pixel = 14
        Metadata complete = true
        Thumbnail series = false

Generating OME-XML (schema version 2015-01)
Parsing schema path
http://www.openmicroscopy.org/Schemas/OME/2015-01/ome.xsd
Validating OME-XML
No validation errors found.

dgault, 2016-11-18:
Hi,

I have tested converting and using the same showinf commands on some CZI files but have not been able to reproduce the same issue just yet. However it does look like similar reports we have seen on Windows systems were the showinf -omexml appears to hang and not output the xml to the display. 

If these are likely large files it might be worth trying to send the output to a text file rather than the cmd display:
showinf -no-upgrade -nopix -omexml-only test.ome.tiff >> "showinf-output.txt" 

David Gault

sebi06, 2016-11-18:
Perfect. This works and is actually even better!

Thanks.

Sebi

closed 2016-11-18 by sebi06