ome/bioformats #2697: Set new options of CZIReader programmatically?

sebi06, 2016-12-14:
Hi guys,

first of all congratulations to the 5.3.0 release. I really happy to see this happening. Nevertheless I have some questions. In the webpage it is stated:

"added two options to the CZI reader to disable autostitching and exclude pyramid file attachments. Added new checkboxes to the CZI configuration interface of the ImageJ plugin to activate these options"

When I open the configuration section(BioFormats Plugins Configuration) related to CZI images, I only see 4 different checkboxes: Enabled, Windowless, Attachment, AutoStitch.

Where is the second option or is this a misunderstanding from my side. AutoStich = active --> Tiling + Pyramid Levels and no active --> no tiling and no pyramid levels?

And is there a way to set this option programmatic? How would I (de)activate this options when I use the MATLAB or Python wrapper from BioFormtas 5.3.0?

Especially those last workflows, where people use wrappers for BioFormats are really important for our users.

Sebi

sbesson, 2016-12-14:
Hi @sebi06,

thanks for the quick feedback. The CZI options available in 5.3.0 should behave as follows:
- if the `zeissczi.autostich` option is set to false (`Autostitch` checkbox disabled), each tile of the largest resolution image will be read as an individual series i.e. similarly to Bio-formats 5.1.x and prior versions. If the options is set to true and applicable, these tiles will be stitched together into a single high-resolution image and each resolution level will also read as a separate image. In the latter case, the flattened resolution API will determine whether these pyramid levels should be treated as separate series (as in ImageJ) or as subresolutions of a pyramidal image (as in OMERO).
- if the `zeissczi.attachment` option is set to true (`Attachement` checkbox enabled), the macro and label images will be read as additional series independently of the autostich option

The table in https://github.com/openmicroscopy/bioformats/pull/2681#pullrequestreview-11999455 should give a summary of the various combinations of options and their effect on the data layout. We will be trying to migrate these scenarios into some proper user documentation as a follow-up of 5.3.0.

We will work on a few code examples of usage of the `DynamicMetadataOptions` class and get back to you. Do you have examples of Python scripts working with Bio-Formats 5.2.x as the migration of python-bioformats to Bio-Formats 5.2.x is still ongoing.

sebi06, 2016-12-15:
Hi,

some example using python-bioformats can be found here:

https://github.com/sebi06/BioFormatsRead

I am wondering if the 5.3.0 still offers the function getResolutionCount that was mentioned earlier. I could not find this any more on: 

http://downloads.openmicroscopy.org/bio-formats/5.3.0/api/loci/formats/meta/MetadataRetrieve.html

And how would I activate those option when using this inside Fiji via Python or the macro language? This is also quite important.

sebi06, 2016-12-15:
I just updated the BioFormtasRead repository and added some stuff to test a CZI with a pyramid with BioFormats 5.1.10 and 5.3.0.

The output is attached.
[5x5_withPyramid_output_BF5.3.0.txt](https://github.com/openmicroscopy/bioformats/files/654281/5x5_withPyramid_output_BF5.3.0.txt)

[5x5_withPyramid_output_BF5.1.10.txt](https://github.com/openmicroscopy/bioformats/files/654279/5x5_withPyramid_output_BF5.1.10.txt)

sbesson, 2016-12-16:
HI @sebi06,

please have a look at [this gist](https://gist.github.com/sbesson/3c9f27de36339b5f2ae03b21599b20d2) which provides two minimal examples covering Fiji and MATLAB.

No API has been dropped between 5.2.4 and 5.3.0. The `getResolutionCount` method you mentioned is part of the [IFormatReader](http://downloads.openmicroscopy.org/bio-formats/5.3.0/api/loci/formats/IFormatReader.html#getResolutionCount--) interface. The MATLAB gist above should also include an example of usage of this API.

Regarding `python-bioformats`, having briefly looked at the code the place that needs to be adapted is the [ImageReader.init_reader()](https://github.com/CellProfiler/python-bioformats/blob/1.0.9/bioformats/formatreader.py#L702) which currently sets the options and initializes the reader. It might be worth engaging a discussion with the CellProfiler team to think of the best way to give more control over these options. Two possibilities coming to my mind would be:

```
rdr = bioformats.ImageReader(imagefile, perform_init=False)
dynoptions = create_dynamic_metadata_options(option1=value1, option2=value)
rdr.setMetadataOptions(dynoptions)
rdr.init_reader()
```

or

```
rdr = bioformats.ImageReader(imagefile, perform_init=False)
dynoptions = create_dynamic_metadata_options(option1=value1, option2=value)
rdr.init_reader(metadataoptions=dynoptions)
```

closed 2017-08-08 by sbesson