ome/bioformats #2763: Problem with CZI and BioFormats inside Python Scripts

sebi06, 2017-02-20:
Hi,

when I use the follwing python script inside Fiji, it return a resolution count = 1. But the resolution count must be = 3 (see boutput for seriescount). I am using BioFormats 5.3.3 but could not figure out so far, how to achieve this. When I open the file manually, I can cleary see the three different pyramid levels.

```
from loci.plugins import BF
from loci.common import Region
from loci.plugins.in import ImporterOptions
from loci.formats import ImageReader
from loci.formats import MetadataTools
from ij import IJ
from ome.units import UNITS
from loci.formats.in import ZeissCZIReader

imagefile = r'c:\Testdata\CZI\5x5_withPyramid.czi'

IJ.log("Current File: " + imagefile)
czireader = ZeissCZIReader()
czireader.allowAutostitching()
omeMeta = MetadataTools.createOMEXMLMetadata()
czireader.setMetadataStore(omeMeta)
czireader.setId(imagefile)
rescount = czireader.getResolutionCount()
seriesCount = czireader.getSeriesCount()  	 
czireader.close()

imageCount = omeMeta.getImageCount()
IJ.log("Resolution Count: " + str(rescount))
IJ.log("Total # of image series (from BF reader): " + str(seriesCount))
IJ.log("Total # of image series (from OME metadata): " + str(imageCount))

options = ImporterOptions()
options.setOpenAllSeries(False)
options.setShowOMEXML(False)
options.setStitchTiles(True)
options.setId(imagefile)

imps = BF.openImagePlus(options)
for imp in imps:
	imp.show()
```

sbesson, 2017-02-20:
Hi @sebi06, it looks like the default is to have resolutions flattened, as you can verify with

```
print czireader.hasFlattenedResolutions()
```

In order to activate the subresolution API, can you try to execute
```
czireader.setFlattenedResolutions(False)
```

before the `setId()` call? This should be sufficient to display the various resolutions.

Best,
Sebastien

sebi06, 2017-02-22:
Hi Sebastien,

if i open the file with the pyramid from the GUI using the normal BioFormats Importer with the AutoStitch option deactivated, I get 25 image series (as expected).
In contrast, when this option is activated, I get 3 Series corresponding to the 3 pyramid levels.

When I try do run my script with setFlattenedResolution(False) or ...(True) i get:

Current File: c:\5x5_withPyramid.czi
hasFlattenedResolution: False
Resolution Count: 3
Total # of image series (from BF reader): 1
Total # of image series (from OME metadata): 1

Current File: c:\CZI\5x5_withPyramid.czi
hasFlattenedResolution: True
Resolution Count: 1
Total # of image series (from BF reader): 3
Total # of image series (from OME metadata): 3

What I am struggeling with is the number of series. Why is this = 3 when there are pyramid levels and 25 if there are no pyramid levels? What am I missing here?

sbesson, 2017-02-22:
Hi @sebi06,

the dimensions you pasted above read in agreement with the modification brought to Bio-Formats 5.3.0  (https://github.com/openmicroscopy/bioformats/issues/2681) and the workflows we discussed in https://github.com/openmicroscopy/bioformats/issues/2634#issuecomment-264697079. From my reading, the three scenarios are:
- if the `zeissczi.autostich` option is deactivated, 25 tiles are unconditionally detected as separate image/series
- if the `zeissczi.autostich` option activated and the `setFlattenedResolution(true)` API is used, the tiles are stitched together into the full-resolution image and overall three pyramidal levels are detected.
As per the flattened resolutions choice, each pyramidal level is effectively detected as a separate image/series and the metadata does not reflect the hierarchy between individual images.
- if the `zeissczi.autostich` option activated and the `setFlattenedResolution(false)` API is used, the same scenario as above with regard to stitching tiles. With the sub-resolution API activated, the pyramidal levels are now properly grouped together and detected as three different resolution level of a single pyramidal image/series.

The main difference between the scripting interface and the graphical interface is that the former gives you access to the full API in particular the `setFlattenedResolutions()` API while the GUI is restricted to the default behavior i.e. flattened resolutions.

Does this answer your question?
Sebastien

sebi06, 2018-03-02:
Hi Sebastien,

I did no have time to check and test this for awhile, but now i tried. And i amstill not really sure if i understand here what is going on.

- zeissczi.autostitch refers to the check inside the plugins configuration windows and its state can be read via czireader.allowAutostitching()?
- How do set this checkbox from a python script?
- The CZIreader gives me always 5 series, but what do I have to use in order to get the number of 120 for this czi (120 Tiles)

CZI: https://www.dropbox.com/s/cm1j7yzumqwedfz/OverViewScan.czi?dl=0

Or sometime  I get czireader.getSeriesCount() = 2 ? But I have no idea why. It should be either 4 (for the 4 Pyramid levels or 120 (no Pyramid and 1 series per Tile

See also the screeshots.

[image]

[image]

[image]

Here is my test script:

```
#@UIService uiService
#@LogService log

# read in and display ImagePlus object(s)
from loci.plugins import BF
from loci.common import Region
from loci.plugins.in import ImporterOptions
from loci.formats import ImageReader
from loci.formats import MetadataTools
from ij import IJ
from ome.units import UNITS
# ZEISS STUFF
from loci.formats.in import ZeissCZIReader

def getImageSeries(imps, series):
    imp = imps[series]
    # get the stack
    imgstack = imp.getImageStack()
    slices = imgstack.getSize()
    width = imgstack.getWidth()
    height = imgstack.getHeight()
    log.info("Show Series #        : " + str(series))
    log.info("ImgStack.getSize()   : " + str(slices))
    log.info("ImgStack.getWidth()  : " + str(width))
    log.info("ImgStack.getHeight() : " + str(height))

    return imp

def getCZIinfo(imagefile,
               showimage=False,
               setflatres=False,
               setreslevel=0,
               openallseries=True,
               showomexml=False,
               stitchtiles=False,
               setconcat=False):

    CZIinfo = {}
    czireader = ZeissCZIReader()
    czireader.setFlattenedResolutions(setflatres)
    czireader.setId(imagefile)
    #czireader.setResolution(setreslevel)
    CZIinfo['rescount'] = czireader.getResolutionCount()
    CZIinfo['seriesCount'] = czireader.getSeriesCount()
    CZIinfo['flatres'] = czireader.hasFlattenedResolutions()
    CZIinfo['getreslevel'] = czireader.getResolution()
    # Dimensions
    CZIinfo['SizeT'] = czireader.getSizeT()
    CZIinfo['SizeZ'] = czireader.getSizeZ()
    CZIinfo['SizeC'] = czireader.getSizeC()
    CZIinfo['SizeX'] = czireader.getSizeX()
    CZIinfo['SizeY'] = czireader.getSizeX()
    # check for autostitching and possibility to read attchmenst
    CZIinfo['AllowAutoStitching'] = czireader.allowAutostitching()
    CZIinfo['CanReadAttachments'] = czireader.canReadAttachments()

    # get OME data
    omeMeta = MetadataTools.createOMEXMLMetadata()
    CZIinfo['imageCount_OME'] = omeMeta.getImageCount()

    log.info("Current Image File : " + imagefile)
    log.info("Get Resolution Count    : " + str(CZIinfo['rescount']))
    log.info("Get Resolution Level    : " + str(CZIinfo['getreslevel']))
    log.info("hasFlattenedResolution  : " + str(CZIinfo['flatres']))
    log.info("# of Series (CZIReader) : " + str(CZIinfo['seriesCount']))
    log.info("# of Images (OMEMeta)   : " + str(CZIinfo['imageCount_OME']))
    log.info("SizeT   : " + str(CZIinfo['SizeT']))
    log.info("SizeZ   : " + str(CZIinfo['SizeZ']))
    log.info("SizeC   : " + str(CZIinfo['SizeC']))
    log.info("SizeX   : " + str(CZIinfo['SizeX']))
    log.info("SizeY   : " + str(CZIinfo['SizeY']))
    log.info("AllowAutoStitching : " + str(CZIinfo['AllowAutoStitching']))
    log.info("CanReadAttachments : " + str(CZIinfo['CanReadAttachments']))

    if showimage:

        # read in and display ImagePlus(es) with arguments
        options = ImporterOptions()
        options.setOpenAllSeries(openallseries)
        options.setShowOMEXML(showomexml)
        options.setStitchTiles(stitchtiles)
        options.setConcatenate(setconcat)
        options.setId(imagefile)

        log.info("setOpenAllSeries   : " + str(openallseries))
        log.info("setShowOMEXML      : " + str(showomexml))
        log.info("setStitchTiles     : " + str(stitchtiles))
        log.info("setConcatenate     : " + str(setconcat))

        # open the ImgPlus
        imps = BF.openImagePlus(options)
        
        for series in range(0, CZIinfo['seriesCount']):
            imp = getImageSeries(imps, series)
            imp.show()

    czireader.close()

    return CZIinfo

# clear the console automatically when not in headless mode
uiService.getDefaultUI().getConsolePane().clear()

# define testfile
imagefile = r'c:\Output\Guided_Acquistion\OverViewScan.czi'
#imagefile = r'c:\Users\m1srh\Documents\Testdata_Zeiss\Castor\EMBL\96well\testwell96_woatt.czi'

info = getCZIinfo(imagefile,
                  showimage=True,
                  setflatres=False,
                  setreslevel=0,
                  openallseries=True,
                  showomexml=False,
                  stitchtiles=False,
                  setconcat=False)

#for x in info.keys():
#    print str(x) +" => " + str(info[x])
```

sbesson, 2018-03-05:
Hi @sebi06 

 >  zeissczi.autostitch refers to the check inside the plugins configuration windows and its state can be read via czireader.allowAutostitching()?

Yes, `zeissczi.autostich` is the reader option which controls whether the Zeiss CZI reader will automatically attempt to stitch tiles if applicable. The `ZeissCZIReader.allowAutostitching()` API returns the value of this option.

>    How do set this checkbox from a python script?

I have turned your Jython code into a Gist - see https://gist.github.com/sbesson/42bcb21dc30f3644c17aabb6f5a1d917 and added a couple of changes on top of it. The last modification shows shows how to consume the Bio-Formats options API to control the tile stitching. This is done in two places:
- before `czireader.setId`, a `DynamicMetadataOptions` object is created, populated with the CZI options and passed to the reader,
- before `BF.openImagePlus`, the options are created using the ImageJ preferences API `Prefs.set(property, value)`. Note although these preferences are applied, they are not refreshed in the UI.

>    The CZIreader gives me always 5 series, but what do I have to use in order to get the number of 120 for this czi (120 Tiles)

With the change above, setting `stitchtiles=False` should be sufficient to return the raw tiles as individual series rather than the auto-stitched images

[image]

> Or sometime I get czireader.getSeriesCount() = 2 ? But I have no idea why. It should be either 4 (for the 4 Pyramid levels or 120 (no Pyramid and 1 series per Tile

In your first screenshot, there are two reasons why the number of series is equal to 2:
- the second series corresponds to the attachment image. In the gist above, I have also used the option API to disable the reading of these attachments using the `zeissczi.attachments` option
- the first series corresponds to a multi-resolution pyramid image as shown by `Resolution Count: 4`. Setting `setflatres` is set to `False` means the `cziReader` will not initialize each pyramidal resolutions as a flattened series but combines all of them as a single multi-resolution OME image. 

Since the Bio-Formats ImageJ API (`BF.openImagePlus`) only supports flattened resolutions, you probably do not want to separately read the dimension with `setFlattenedResolutions(false)` otherwise you might end up with dimensionality mismatches.

Hope this answers all your questions,
Sebastien

sebi06, 2018-03-13:
Many THx. Now it works for me, even if this is not really what I would call straight foreward.

Sebi

closed 2018-03-13 by sebi06