ome/bioformats #2775: Please add inputstream/outputstream as input/output options for IFormatReader in LociTools...

markehammons, 2017-03-04:
Hi All,

Right now it appears that when using locitools to read or write a new file, I need to give it a filename using the setId function. I would prefer to be able to send the output to a remote server, using an output stream. Could you support the InputStream/OutputStream classes as inputs to an IFormatReader/IFormatWriter?

https://downloads.openmicroscopy.org/bio-formats/5.1.2/api/loci/formats/IFormatReader.html

dgault, 2017-03-06:
Hi,

One way that you could potentially achieve this functionality would be using `Location.mapFile` - https://github.com/ome/ome-common-java/blob/master/src/main/java/loci/common/Location.java#L219-L224

Typically an id is a filename on disk but it is possible to have this mapped to an IRandomAccess object instead.

A full example of reading and writing in this manner can be found here: https://github.com/openmicroscopy/bioformats/blob/develop/components/formats-gpl/utils/ReadWriteInMemory.java

Hopefully that will be suitable for your needs but if you have any further questions please do let us know.

sbesson, 2017-04-12:
Assuming https://github.com/openmicroscopy/bioformats/issues/2775#issuecomment-284415638 is an acceptable workaroung for the time being, closing this since the in-memory example is now reviewed, tested and better exposed in the documentation with Bio-Formats 5.4.1 - see https://github.com/openmicroscopy/bioformats/pull/2812.

closed 2017-04-12 by sbesson