ome/bioformats #2791: Image Attachment of CZI is not read correctly with BioFormats 5.3.4

sebi06, 2017-03-20:
Hi,

some of our systems use a preview camera to acquire an overview image of the sample and/or carrier to use this for sample or sample carrier recognition. This image is stored as an attachment inside the CZI. When the respective option is activated using BioFormats 5.3.4 this image is read but only partially (see attached screenshots)

Maybe you read only 1 series of it? The image actually consist out of a few single images and is "tiles" together. Let me know if you need more detailed info or more testfiles to fix this

The original CZI image can be found here:

https://drive.google.com/open?id=0B8YXT9fyYIBzWEg5ZTl5ZDlQTDg

[image]
[image]

dgault, 2017-03-20:
Thank you for providing a sample file to test this with. I was able to reproduce the issue as described with the latest Bio-Formats build. I have added the details to card on the Bio-Formats Trello board to be prioritised.

melissalinkert, 2018-12-19:
For reference, the Trello card is: https://trello.com/c/33ZpzJ9G/122-zeiss-czi-image-attachment-incorrect-width

zeissmicroscopy, 2019-02-12:
I recreated the link with an example of an CZI with attachment:

https://drive.google.com/open?id=1LsM94JW2l_B9ElcxNlSVXLoztJwE6_kJ

melissalinkert, 2019-02-12:
Thank you, @zeissmicroscopy.  The file has now been downloaded and copied to our testing repository (```inbox/gh-2791/```).

melissalinkert, 2021-04-19:
Testing this file with https://github.com/ome/bioformats/pull/3681, ```showinf -option zeissczi.trim_dimensions true``` results in the correct image size.

melissalinkert, 2022-08-29:
This was fixed in 6.7.0, as a result of the changes in https://github.com/ome/bioformats/pull/3687.

closed 2022-08-29 by melissalinkert