ome/bioformats #2798: TIFF to TIFF conversion is corrupted

felixveysseyre, 2017-03-22:
Hello,

I am using `bfconvert` executable to extract images from a TIFF file:
`bfconvert -bigtiff -series 0 "input.tiff" ".../output.%z.%t.%c.tiff" `

The process run correctly but at the end the TIFF image `output.0.0.0.tiff` seems corrupted:
[image]

[Here](https://drive.google.com/file/d/0B-xvT4_QuWKMamhtRkZINnBQWGM/view?usp=sharing) is the data for which I got the issue.

I am using `bfconvert 5.4.0`.

Thanks for your time and your tools !

dgault, 2017-03-23:
Hi Felix,

I have been testing this morning with the sample file provided and the command line tools from Bio-Formats 5.4.0. So far I have unable to recreate the same 'noise' as you are seeing it.

A couple of basic questions I would have to start:
- Can you confirm that running `bfconvert -version` displays `Version: 5.4.0`
- What operating system were you using?
- What software were you using to display the converted image?
- Would it be possible to upload the converted file with the noise for comparison?

felixveysseyre, 2017-03-24:
Hello David,

I can confirm I am using latest `bftools` executables:

> Version: 5.4.0
> Build date: 20 March 2017

But the same error was also occurring with `5.3.3`.

I have tested this behavior on both Ubuntu (14.04) and MacOS (10.11).

I am not able to display directly the `output.0.0.0.tiff` file. I tried :
- Ubuntu: Default viewer
- Mac: Default viewer, GIMP
- Windows: Default viewer, GIMP

Do you advise any viewer for MacOS or Ubuntu ?

The `output.0.0.0.tiff` file is later tiled into multiple JPEG files thanks to a C++ executable built on top of ITK and SCIFIO. It corresponds to the image I uploaded for this issue.

Since I was not able to open the `output.0.0.0.tiff`, file I supposed it came from `bfconfert`.

Thanks for your time !

felixveysseyre, 2017-04-06:
@dgault Do you any news ?

dgault, 2017-04-07:
Hi Felix,

Firstly apologies for the delayed response, the `output.0.0.0.tiff` would indeed be the file generated from the bfconvert command you were using. A simple way to view this output would be to use the `showinf` command which is available along with bfconvert as part of the Bio-Formats command line tools .

`showinf output.0.0.0.tiff` will read and print the metadata from the output file and an image viewer will open. Due to the filesize this may be quite memory intensive. 

Another option would FIJI (https://fiji.sc/) which comes with Bio-Formats included. Once installed you can read files using `Plugins -> Bio-Formats -> Bio-Formats Importer`

sbesson, 2018-03-14:
This issue has been addressed and has no further activity. It is therefore being closed. Thank you
for your contributions, they are much appreciated.

closed 2018-03-14 by sbesson