ome/bioformats #2803: bfconvert and CZI question regarding the image attachments

sebi06, 2017-03-28:
Hi guys,

I use a batch script to run **bfconvert** on all CZI files inside a folder.
```
@REM Run the bfconvert.bat for all CZI files inside the directory
@REM for /r %%f in (*.czi) do call bfconvert.bat -no-upgrade %%f %%~nf.ome.tiff
if "%EXPORTOMETIFF%" == "-export" (
	for /r %%f in (*.czi) do (
		@ECHO File2Convert: "%%f"
		@ECHO File2Save: "%%~nf.ome.tif"
		call %BFCONVERT_PATH% -no-upgrade "%%f" "%%~nf.ome.tiff"
	)
)
```

I assume that reading the CZIs (via CZIReader) is the first step required here. In Fiji for instance it is possible to disable reading the image attachments, but so far I have no idea on how to disable this when using **bfconvert**.

The workflow problem is as follows:

- acquire 1 image per well  in a 96 well --> one single CZI including one image attachment
- CZIs are splitted into 96 single CZIs where now every single CZI also has the attachment inside
- use BioFormats with Attachment option disabled --> one gets 96 images --> Perfect
- use **bfconvert** --> 96 OME-TIFFs where every OME-TIFF has the attachment built-in

When I know move to CellProfiler, KNIME, etc. all those SW have issues to deal with this additional image series (Attachment) which is useless for image analysis anyway. Therefor it would be nice to be able to remove those attachments using **bfconvert** already and avoid all those downstream issues.

simleo, 2017-03-29:
Hi,

since Bio-Formats 5.4.0 you can add key/value pair options to `bfconvert`. One of the supported options is `zeissczi.attachments`, which defaults to `true`. You should upgrade to the latest Bio-Formats version (if you haven't done so already) and try running `bfconvert` with `-option zeissczi.attachments false`, e.g.:

```
bfconvert -option zeissczi.attachments false foo.czi foo.ome.tif
```

This should prevent attachment images from being read.

sebi06, 2017-03-30:
Perfect. Is there an official documentation somewhere where I can find all those options. I did look for it but was not able to find those options.
Is there then also an option to only load the full resolution pyramid level from CZI files and an option to tile / not tile the images? (Both is possible within Bioformats itself)

sbesson, 2017-03-30:
Hi @sebi06,

thanks for your feedback.  As of Bio-Formats 5.4.0, the following reader options should be available:

Format | Option | Default | Description
-------|--------|---------|------------
Zeiss CZI | zeissczi.attachments | true | Include attachment images
Zeiss CZI | zeissczi.autostitch | true | Automatically stitch tiled images
Nikon ND2 | nativend2.chunkmap | true | Use chunkmap table to read image offsets
Leica LIF | leicalif.old_physical_size | false | Ensure physical pixel sizes are compatible with versions <= 5.3.2

We do not have have a documentationpage listing all these options yet but this had been on our todo list. Now that OMERO 5.3.0 is out, we will certainly prioritize this addition. /cc @hflynn

For the loading of the full resolution image only, you can achieve this by using the [-series option](https://www.openmicroscopy.org/site/support/bio-formats5.4/users/comlinetools/conversion.html#cmdoption-bfconvert-series) of the bfconvert/showinf utility e.g.:

```
bfconvert -option zeissczi.autostitch true -series 0 largeimage.czi fullresolution.ome.tiff
```

sebi06, 2017-03-31:
Hi Sebastien,

that really helps. And what is the right option to read only the full resolution image with a series with multiple scenes?

Example:
- Acquistion does 3 different  10x10 TileRegions
- Every Scene (TileRegion) has now 1 Series with FullResulution and 2 more Pyramid levels
- Alltogether this leads to 3 * (1 +2) = 9 Series

Series 0 - FullRes 1st Region
Series 1 - Pyramid2 1st Region
Series 2 - Pyramid2 1st Region
Series 3 - FullRes 2nd Region
Series 4 - Pyramid2 2nd Region
Series 5 - Pyramid2 2nd Region
Series 6 - FullRes 3rd Region
Series 7 - Pyramid2 3rd Region
Series 8 - Pyramid2 4rd Region

So in order to read out all 3 Tile Regions at FullRes I would need to adress (for the example) Series 0, 3, 6. In conjuction with the series option I can only pass one series number, right? And I would have to figure out which series actually are the one having the FullResolution?

So I am womdering if there is an easy option to read all Regions in FullResolution or using a distinct Pyramid Level? Or is this not a UseCase from your point of view?

sbesson, 2017-03-31:
Hi @sebi06, 

briefly coming back to the options table, just a heads up that we are current working on our documentation (see https://github.com/openmicroscopy/bioformats/pull/2804) and expect to have it delivered in Bio-Formats 5.4.1.

Regarding your conversion scenario, unfortunately there is no easy way to achieve your workflow directly with the `bfconvert` utility without the caveats you mentioned i.e. knowing which series contain the full resolution (or pyramidal level) of each region and invoking the command multiple times as you said.

For your use case, it should be possible to derive a custom converter (for instance based on the example in https://www.openmicroscopy.org/site/support/bio-formats5.4/developers/conversion.html) with the following additions:

#. create a `DynamicMetadataOptions` object, set the CZI options to exclude attachments and pass it to the reader using `setMetadataOptions()`
#. activate the subresolution API using `setFlattenedResolutions(false)` to loop over regions/scenes rather than individual pyramidal levels
#. after initializing the readers, loop over the different regions using `setSeries(int)`
#. for each region, select the desired resolution level using `setResolution(int)`
#. read the planes and convert into the new file as previously

Closing this issue as the initial problem (image attachment option and documentation) is resolved. We can use another topic/issue to discuss the conversion of individual resolution levels for each scene.

closed 2017-03-31 by sbesson