ome/bioformats #2805: bfconvert and spitting patterns

sebi06, 2017-03-31:
Hi guys,

I have the following CZI that contains 2 timepoints:

https://www.dropbox.com/s/dgkm4g7qd5za48s/Test2%20ABC-Scene-1-P1-A01.czi?dl=0

Now I want to convert this file to OME-TIFF and split it into single timepoints using BF5.4.0:

```
c:\bftools>bfconvert -no-upgrade "Test2 ABC-Scene-1-P1-A01.czi" "Test2 ABC-Scene-1-P1-A01_T%%t_.czi"
c:\bftools\Test2 ABC-Scene-1-P1-A01.czi
ZeissCZIReader initilaizing c:\bftools\Test2 ABC-Scene-1-P1-A01.czi
[Zeiss CZI] -> c:\bftools\Test2 ABC-Scene-1-P1-A01_T%t.ome.tiff [OME-TIFF]
      Converted 2/2 planes (100%)
[done]
```
As a result I get as expected 2 OME-TIFF files:

Test2 ABC-Scene-1-P1-A01_T0.ome.tiff
Test2 ABC-Scene-1-P1-A01_T1.ome.tiff

But when I open one of those OME-TIFF it still has 2 Timepoints? Any idea?

dgault, 2017-03-31:
Hi @sebi06,

I was able to reproduce what you were seeing but this appears to be expected behaviour. Each time-point has been written to a separate file however upon import the files are grouped together as a single dataset and as such both files are imported together. From the OME-XML you will be able to see reference to the individual files for each timepoint (in this case I have called them `exportTest`):

```
<TiffData FirstC="0" FirstT="0" FirstZ="0" IFD="0" PlaneCount="1">
<UUID FileName="exportTest_T0.ome.tif">urn:uuid:2a88f1fb-0238-40b2-991e-a4c53e8c71d5</UUID>
<TiffData FirstC="0" FirstT="1" FirstZ="0" IFD="0" PlaneCount="1">
<UUID FileName="exportTest_T1.ome.tif">urn:uuid:dbad5672-5822-47d1-a636-0338e66db5c2</UUID>
```

sebi06, 2017-03-31:
Ok, so maybe I got confused by the quite similar files sizes of the results. The actual single files contain only the image data (pixels) from one timpoint but every file has metadata ontop. 

And because of the metadada Fiji' BioFormats imports both files automatically, even if i check the open to treat every file as an individual file. Is there any way to read those wothout the automatic grouping. I did not find anything. The "Open files individually" option did not work.

[image]

The idea behind splitting those files was actually to get the single file <4GB since some software packages have still issues with this.

So if the "other" SW has no clue about the metadata (and therefore no automatic grouping) than it will import the individual small files. Is this correct?

dgault, 2017-04-03:
The issue will be around how the metadata is being parsed and used by the software. Splitting the files using this method will result in a single dataset split over multiple files with associated metadata in each file. The alternative would be to use the new companion file option which creates a single companion file for the metadata and a smaller stub to reference it in each file: `bfconvert -option ometiff.companion outputFile.companion.ome inputFile.tiff outputFile.ome.tiff`

You can read more about multi-file OME-TIFF here: https://www.openmicroscopy.org/site/support/ome-model/ome-tiff/specification.html#multi-file-ome-tiff 

When using the option to open files individually the Bio-Formats importer will check with the reader to retrieve the file grouping options for that file, in either of the above scenarios the OME-TIFF reader will notify that the files must grouped and therefore it always opens them as a single dataset.

Using the Bio-Formats importer, what you would have to do would be to open a subset using the range option. So for external software it would be down to the implementation to decide how to handle this scenario.

sbesson, 2018-03-14:
This issue has been addressed and has no further activity. It is therefore being closed. Thank you  for your contributions, they are much appreciated.

closed 2018-03-14 by sbesson