ome/bioformats #2808: NiftiReader fails to read multi-channel images (start of a fix included)

bogovicj, 2017-03-31:
This relates to [a ticket](http://trac.openmicroscopy.org/ome/ticket/13291#ticket) I submitted last September. I've observed that bioformats interprets 4d multi-channel nifti images as 3d.

After some digging, I seem to have found part of the problem. https://github.com/bogovicj/bioformats/commit/6ebb68fd63b76c41b5ed423d115f2678e344d0ef enables the bioformats `NiftiReader` to more correctly read the nifti image metadata.

Installing the above patch into Fiji and using the bioformats reader through the GUI yields an ImagePlus with **correct dimensions, but incorrect data**.  This is consistent with the fact that 
`BF.openImagePlus( path )` now yields an image with correct dimensions, but wrong data.

It's puzzling me, because [this snippet](https://github.com/bogovicj/bioformats/blob/e8a73414424e76c0b92b5fce6c00efe756c3cd08/components/bio-formats-plugins/utils/OpenNifti.java#L28-L50) gives something with the **correct size and correct data**.  I'm hoping that someone with more experience can help resolve the last piece of this bug.

dgault, 2017-04-03:
Hi John,

I was able to test your commit with the original sample file you had provided previously. I believe the issue with the data is due to the expected interleaving of the channel data. On the below line it is assumed that if the size of C is greater than 1 then the image is RGB and if its RGB then it will be interleaved data. 

https://github.com/bogovicj/bioformats/commit/6ebb68fd63b76c41b5ed423d115f2678e344d0ef#diff-1696608d5491b8873c77d696b4f4cc4aR274 

In the case of the sample file you provided the image data does not appear to be interleaved and setting this to false should display the data correctly.

The reason for the original interleaved check was for 2 specific pixel types [NIFTI_TYPE_RGB24](https://github.com/bogovicj/bioformats/commit/6ebb68fd63b76c41b5ed423d115f2678e344d0ef#diff-1696608d5491b8873c77d696b4f4cc4aR328) and [NIFTI_TYPE_RGBA32 ](https://github.com/bogovicj/bioformats/commit/6ebb68fd63b76c41b5ed423d115f2678e344d0ef#diff-1696608d5491b8873c77d696b4f4cc4aR340), which is a different scenario that when reading from the extra dimension field.

In short you will have to put a check in place so the original logic for interleaving remains only for the instances of those pixel types.

With that small change this looks like a fix that would be good to integrate back into the Bio-Formats mainline. Would you be interested in opening a PR against the Bio-Formats develop branch once this is resolved?

bogovicj, 2017-04-03:
Thanks @dgault,
for both the feedback and looking into what the issue might be.  The interleaving sure sounds like it could be the cause.

I'll try this out when I have a chance (though it may not be soon), and if it looks good I'll certainly file a PR.