ome/bioformats #2811: SCN to JPEG transform is failling

YOUNGING, 2017-04-05:
I'm using bfconvert command to transform Leica SCN images, the original image is like this (using ImageScope to view)
[image]

but the output image is not what I'm expecting. All the white background become pink, and the tissue become black and blur.

[image]

Can anyone tell me what is the problem here? Thanks a lot.

dgault, 2017-04-05:
Hi @YOUNGING,

I was able to run some tests on existing sample files we have and could not reproduce the issue with the latest Bio-Formats 5.4.0. 

Can you tell us which version you are currently using `bfconvert -version`
Also would it be possible to upload a sample file which produces this issue to the link below:
https://www.openmicroscopy.org/qa2/qa/upload/

YOUNGING, 2017-04-06:
Thanks for your help. I update the bftools  to the latest version (5.4.0), but the issue is still there. I have uploaded one sample file to that link. The filename is 132223-12_022a1963-b32e-467f-8242-b51a5295951f.SCN.

dgault, 2017-04-06:
Thank you for submitting the sample file (QA-17671), I was able to reproduce the issue with the file provided.

This looks like it is potentially a bug in the LeicaSCNReader, comparing the imported image to that when using ImageScope the metadata being read appears to match the expected values however the pixel counts are higher than expected and are being capped at 255 (max value for uint8 data). 

This will require further investigation and debugging on our part.

YOUNGING, 2017-04-08:
@dgault 

I notice that Bio-formats have openslide as a component. Is it possible to use openslide as the default reader for SCN files?

I'm new to Java. If there is anything else I can do to accelerate the debug process, please tell me.

dgault, 2017-04-10:
@YOUNGING  Apologies for the delayed response, I was off at the end of last week. I don't believe the issue is related to  the `dissimilar main images` and we are not directly using openslide for this reader.

The main thing that would be of help to us would be to have more sample files. Have you noticed this issue occurring with a large number of images? Have you noticed any particular settings which cause images to display this problem?

YOUNGING, 2017-04-11:
I have more than 100 images of this format. They all display this problem.

dgault, 2017-04-11:
I have noticed that if I split the channels and open them individually that 2 of them appear to have the 'saturated' effect which is causing the pink in the combined display while one of the channels appears to be displaying as expected.

[image]

Would you have expected to see that from the individual channels or do the 2 brighter channels look wrong?

petebankhead, 2018-01-19:
I saw similar behavior some years ago when directly accessing JPEG-compressed tiles from an SVS file.  Not sure if it's the solution, but the colors here also become more plausible if the photometric interpretation is changed.

I posted a Groovy script for QuPath that shows what happens. Running it requires Bio-Formats, [QuPath](http://qupath.github.io), and a [QuPath extension](https://github.com/qupath/qupath-bioformats-extension) to link the two:
https://github.com/qupath/qupath/issues/141#issuecomment-358951720

rleigh-codelibre, 2018-01-22:
Using the `tiffgt` opengl viewer (from libtiff), this shows the image as expected with the correct colours.  Forcing this to use PHOTOMETRIC=RGB shows different colours than above though.  It looks like the bug is in bio-formats, but I've not yet found why it happens, and I'm unsure it's due to PHOTOMETRIC!=YCbCr; it might be more subtle than that.

petebankhead, 2018-08-02:
As the reference above shows, this resurfaced again with QuPath and so I investigated a bit more.

I also checked with `tiffgt` and see that it does give some odd colors if I force it to PHOTOMETRIC=RGB. However, the _correct_ colors from tiffgt seem to match the colors I get when I run my messy script that artificially forces the PHOTOMETRIC_INTERPRETATION value used by Bio-Formats to RGB.  These colors also look like those I see if I open the image with Aperio ImageScope (from Leica, so presumably correct).

Therefore I continue to be suspicious that YCbCr is somehow involved.

Following the code, it seems that when `CodecOptions.ycbcr == true` then additional processing is applied after JPEG decompression [here](https://github.com/ome/ome-codecs/blob/master/src/main/java/ome/codecs/JPEGCodec.java#L126).

This occurs within [`TiffParser.java`](https://github.com/openmicroscopy/bioformats/blob/0595010eecc58334c37b1dd76a0a173ed00d2ef6/components/formats-bsd/src/loci/formats/tiff/TiffParser.java#L784) whenever
```java
codecOptions.ycbcr =
      ifd.getPhotometricInterpretation() == PhotoInterp.Y_CB_CR &&
      ifd.getIFDIntValue(IFD.Y_CB_CR_SUB_SAMPLING) == 1 && ycbcrCorrection;
```

Some SCN files work (including the ones from OpenSlide, also within the Bio-Formats downloads), while others appear pink.  In all cases the PHOTOMETRIC_INTERPRETATION is YCbCr, however it seems that:
* for the ones that work, `ifd.getIFDIntValue(IFD.Y_CB_CR_SUB_SAMPLING) == 2` and so `codecOptions.ycbcr == false`
* for the ones that turn pink, `ifd.getIFDIntValue(IFD.Y_CB_CR_SUB_SAMPLING) == 1` and therefore `codecOptions.ycbcr == true`

Running my script has the effect of `codecOptions.ycbcr` always being false (albeit for the wrong reasons); for the pink images this seems to 'correct' them, and for the others it makes no apparent difference.

Therefore it looks like `codecOptions.ycbcr` being true when it comes to performing the JPEG decompression is somehow responsible.

I don't have any example images of my own, although some were posted on the QuPath threads referencing this one. A small additional observation: the latest SCN files appear to also include channel names and colors in the XML, and also
```
<supplementalImage type="label" ifd="6" />
```
There is indeed a label there, but I can't find it as a series in Bio-Formats.

dgault, 2018-08-06:
Thanks for the further investigation @petebankhead, i've been starting to take a closer look at this again today but have not yet got to bottom of exactly what is going wrong in this instance.The extra channel names and color info is something we can work on adding if it is not currently being parsed.

sbesson, 2018-12-11:
Retested with the latest stable release of Bio-Formats, 5.9.4, I confirm the initial faulty behavior.

From the file content, it looks like a Versa dataset:
 
```
sbesson@ls30630:Desktop $ tiffinfo 132223-12_022a1963-b32e-467f-8242-b51a5295951f.SCN | grep model
      <device model="Versa" version="1.0" />
      <device model="Versa" version="1.0" />
```

Retesting with [Bio-Formats 6.0.0-m3](http://downloads.openmicroscopy.org/bio-formats/6.0.0-m3/) which includes the Versa improvements from #3240, the image now opens correctly without the pink background issue reported above. Assuming all other Leica SCN samples mentioned [above](https://github.com/openmicroscopy/bioformats/issues/2811#issuecomment-293134444) are similar, I suspect the upcoming Bio-Formats 6 will fully resolve this issue.

closed 2018-12-11 by sbesson