ome/bioformats #2820: ND2 files cannot be read and appears corrupt since version 5.0.3

SarenT, 2017-04-20:
Hi,

I am trying to open ND2 files from Nikon to save them as TIFFs. But it appears to be with some files, that the file cannot be read properly and looks corrupt (looks like noise only with weird unexpected shapes). I mainly use FiJi but because of this issue I have tried several version from the website. I have limited the issue to version 5.0.3 and above. Here is an example of the same file displayed completely differently:
[image]

I can provice more information about the issue on demand. I am using Ubuntu 16.04 but the issue also occurs on Windows with ImageJ. In the syslog I see following errors:

[bio-formats_error.txt](https://github.com/openmicroscopy/bioformats/files/943427/bio-formats_error.txt)

SarenT, 2017-04-20:
Yes, first I tried out 5.4.1. Then I tried to find out where it started to not to open it correctly.
I tried with and without chunkmap and with and without native library. None of them works with the current release. I will upload the file ASAP. The file is created by NIS elements software.

SarenT, 2017-04-21:
I have uploaded the file as "sample.nd2" right now.

dgault, 2017-04-24:
Hi @SarenT, thank you for uploading the sample file. I was able to test and reproduce the issue you described. From further debugging this appears to be due to an incorrect PixelType being used after parsing the file. I have opened a PR https://github.com/openmicroscopy/bioformats/pull/2823 in order to test a potential fix for issue.

sbesson, 2017-04-27:
@SarenT: thanks again for contacting us, we expect David's fix to be part of the next release of Bio-Formats. In order to catch potential regressions, we will add the sample file you uploaded to us to our data repository for daily testing. One question, could this data be made public under the CC-BY license alongside other [public ND2 images](http://downloads.openmicroscopy.org/images/ND2/) or should it remain private? If needed, we can provide a location to review this file.

SarenT, 2017-04-27:
@sbesson: Sure, it is nice to hear that I could somehow contribute to the project. Could you pleace keep the file private for now. I need to ask other people, who has part in it (although this should not be any problem). Maybe tomorrow I can even provide another file, which can be made public without consent of others.

sbesson, 2017-05-01:
Closing this issue as https://github.com/openmicroscopy/bioformats/issues/2823 is merged

closed 2017-05-01 by sbesson

SarenT, 2017-05-09:
I got an update on my FijI for bioformats (5.5.0) and it can open the sample file I sent you. Thanks a lot!

itayoz, 2020-12-30:
hi guys,
i recently started to use fiji. i took some photos with nikon smz 25 and saved them as nd2 files. i tried to open them with fiji and some opened good, and some open corrupted. how can I fix it?[
[image]