ome/bioformats #2847: Extracting sub images from a TIFF file results in an error

felixveysseyre, 2017-05-10:
Hello,

I am using _bfconvert_ executable to extract images from a TIFF file:

> bfconvert -bigtiff -series 0 "input.tiff" ".../output.%z.%t.%c.tiff"

The process fails quickly with the following error:
> TiffDelegateReader initializing input.tiff
> Reading IFDs
> Populating metadata
> Checking comment style
> Populating OME metadata
> [Tagged Image File Format] -> output.%z.%t.%c.tiff [Tagged Image File Format]
> Exception in thread "main" java.lang.IndexOutOfBoundsException: Index: 1, Size: 1
> 	at java.util.ArrayList.rangeCheck(ArrayList.java:635)
> 	at java.util.ArrayList.get(ArrayList.java:411)
> 	at ome.xml.model.Pixels.getChannel(Pixels.java:628)
> 	at ome.xml.meta.OMEXMLMetadataImpl.getChannelName(OMEXMLMetadataImpl.java:1211)
> 	at loci.formats.FormatTools.getFilename(FormatTools.java:1138)
> 	at loci.formats.FormatTools.getFilename(FormatTools.java:1101)
> 	at loci.formats.tools.ImageConverter.testConvert(ImageConverter.java:541)
> 	at loci.formats.tools.ImageConverter.main(ImageConverter.java:884)

I am using the latest version of BFTools:
> $ ~/Softwares/bftools/bfconvert -version
> Version: 5.5.0
> Build date: 8 May 2017
> VCS revision: d65f35a268ce2b23e13713d58ba24c4a18f3520c

I would like to add the image to this issue but it contains some non anonymized information. What is the correct procedure ?

Thanks for your time !

melissalinkert, 2017-05-10:
Thank you for reporting this problem, @felixveysseyre.  The best way to send the image file is to upload it to http://qa.openmicroscopy.org.uk/qa/upload/.  The OME team can then see it for testing, but it will not become publicly available without your permission.

felixveysseyre, 2017-05-11:
@melissalinkert Thanks for your help !

I just uploaded the image, the id was 17718.

dgault, 2017-05-11:
Thanks @felixveysseyre for submitting the sample file. I was able to test and reproduce the same error that you reported. From debugging the issue I believe I have found the root cause and should have a PR open shortly to put a fix in place.

felixveysseyre, 2017-05-12:
@dgault Thanks a lot for your time !

Could you ping me when the changes have been applied to the release version ?

dgault, 2017-05-12:
Sure thing @felixveysseyre. I have opened the PR now, once this has been tested and merged  it will be scheduled for an upcoming release. I will keep this GitHub issue up to date once the PR has been accepted and once it has been included in a release build.

sbesson, 2017-05-15:
Closing as per #2848. The fix should be available in the next patch release of Bio-Formats.

closed 2017-05-15 by sbesson